MarlinFirmware/Marlin #12388: [bugfix-2.0.x]  BLtouch 'touch' my bed control output

mr-miky, 2018-11-10:
It's been a while since I noticed it but I temporarily ignored it because I needed to complete HW debugging.

In summary it happens that bltouch goes to alter the state of the pin that controls the heater of the bed. At the time I have circumvented the problem by changing the pin that controls the bed.
It always seemed strange to me because all happens on PA0  a.k.a. pin 0 ...

Now that I'm near the finish line with the debugger and I'm trying to figure out what really happens.
I added a brekpoint that monitor **pin** on servo class.
The strange thing is that servo_init ()  correctly set the servo pin and then disconnect it as expected, then from the next servo_probe_init () the servo control **pin** becomes pin 0 / PA0.
servo_probe_init (), through an infinite series of MACRO, call servo.move (...)

I do not understand the following:

From HAL/shared/servo.cpp

```
 *
 * Servo - Class for manipulating servo motors connected to Arduino pins.
 *
 * **attach(pin)           - Attach a servo motor to an i/o pin.**
 * attach(pin, min, max) - Attach to a pin, setting min and max values in microseconds
 *                         Default min is 544, max is 2400

```

```
void Servo::move(const int value) {
  constexpr uint16_t servo_delay[] = SERVO_DELAY;
  static_assert(COUNT(servo_delay) == NUM_SERVOS, "SERVO_DELAY must be an array NUM_SERVOS long.");
  if (this->attach(0) >= 0) {
    this->write(value);
    safe_delay(servo_delay[this->servoIndex]);
    #if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE)
      this->detach();
    #endif
  }
}

```

and in particular

**this-> attach (0)**

There are two possibilities: 
  ° I do not understand
  °  Nobody uses a servo / bltouch.
The first one is more probable and that's why I ask here ...

mr-miky, 2018-11-10:
GRRRR!!! Again !!!!

Yet another flaw of STM32_GENERIC !!!

I did not realize that servo.cpp is a library implemented in STM32_GENERIC  :-(

jmz52, 2018-12-10:
@mr-miky, second option is a correct one (Nobody uses a servo / bltouch)
Servo's code is broken (probably an uncompleted port from some HAL_TEENSY35_36)
See if you can re-use code from HAL_STM32F1/HAL_Servo_Stm32f1.cpp.

If I recall correctly problem root cause was '#ifdef \_SERVO_H_' used in both module\servo.h from Marlin and libraries\Servo\src\Servo.h from stm32duino. Marlin defines \_SERVO_H_ first, effectively blocking library from stm32duino.
HAL_Servo_Stm32f1.cpp was modified to work without Servo library from stm32duino.

boelle, 2019-02-21:
@mr-miky problem solved?

mr-miky, 2019-02-22:
I made a patch that honestly is really horrible more than anything else to be able to finish the hardware test of my custom control board. I would like to find the time to fix it but I am very busy

boelle, 2019-02-22:
@mr-miky submit pr to bugfix 2.0 and close this one

minimum close this one

mr-miky, 2019-02-24:
Actually the problem in the source exists and closing the issue does not make the problem disappear.

AnHardt, 2019-02-26:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/shared/servo.cpp#L98-L99

is the key to your question.
`this-> attach (0)` does not mean "attach pin 0", but "attach the already known and stored in`this->servoIndex` pin". If we do know the right servo object we already have the right pin.

mr-miky, 2019-02-27:
@AnHardt 
I understood the mechanism but I suspect the stm32generic 'servo' library interprets it differently.
the problem is that attach (0) goes to alter the state of pin PA0.

from https://github.com/danieleff/STM32GENERIC/blob/master/STM32/libraries/Servo/src/Servo.cpp

`uint8_t Servo::attach(int pin) {
    this->pin = pin;
    this->minAngle = 0;
    this->maxAngle = 180;
    return 1;
}`

mr-miky, 2019-02-27:
It seems to me that that attach (0) brings more incompatibility than benefits.
`if (pin > 0) servo_info[this->servoIndex].Pin.nbr = pin;`
In the end it only avoids making a further pin assignment.

mr-miky, 2019-02-28:
@AnHardt   
"This implies: servos can't be connected to `PA0`." 
Perhaps you meant **nothing** can be connected to PA0

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

mr-miky, 2019-06-26:
I close it so you don't get bored but the problem still exists

closed 2019-06-26 by mr-miky