MarlinFirmware/Marlin #12390: [bugfix-2.0.x] Error compiling lcdprint_u8g.cpp redefinition of struct _uxg_fontinfo_t

brucehvn, 2018-11-10:
### Description

Downloaded the latest bugfix-2.0.x code on 10-Nov-2018.  Rebased to get my config changes applied and tried to compile.  I get the following output from the Arduino IDE:
```
"/home/bruce/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/bruce/.arduino15/packages/arduino/hardware/avr/1.6.20/cores/arduino" "-I/home/bruce/.arduino15/packages/arduino/hardware/avr/1.6.20/variants/mega" "-I/home/bruce/Arduino/libraries/U8glib/src" "/tmp/arduino_build_85356/sketch/src/lcd/lcdprint_u8g.cpp" -o "/tmp/arduino_build_85356/sketch/src/lcd/lcdprint_u8g.cpp.o"
In file included from /tmp/arduino_build_85356/sketch/src/lcd/lcdprint.h:16:0,
                 from /tmp/arduino_build_85356/sketch/src/lcd/lcdprint_u8g.cpp:24:
/tmp/arduino_build_85356/sketch/src/lcd/dogm/u8g_fontutf8.h:19:16: error: redefinition of 'struct _uxg_fontinfo_t'
 typedef struct _uxg_fontinfo_t {
                ^
In file included from /tmp/arduino_build_85356/sketch/src/lcd/lcdprint_u8g.cpp:23:0:
/tmp/arduino_build_85356/sketch/src/lcd/u8g_fontutf8.h:20:16: error: previous definition of 'struct _uxg_fontinfo_t'
 typedef struct _uxg_fontinfo_t {
                ^
In file included from /tmp/arduino_build_85356/sketch/src/lcd/lcdprint.h:16:0,
                 from /tmp/arduino_build_85356/sketch/src/lcd/lcdprint_u8g.cpp:24:
/tmp/arduino_build_85356/sketch/src/lcd/dogm/u8g_fontutf8.h:25:17: error: invalid type in declaration before ';' token
 } uxg_fontinfo_t;
                 ^
/tmp/arduino_build_85356/sketch/src/lcd/dogm/u8g_fontutf8.h:25:17: error: conflicting declaration 'typedef int uxg_fontinfo_t'
In file included from /tmp/arduino_build_85356/sketch/src/lcd/lcdprint_u8g.cpp:23:0:
/tmp/arduino_build_85356/sketch/src/lcd/u8g_fontutf8.h:26:3: note: previous declaration as 'typedef struct _uxg_fontinfo_t uxg_fontinfo_t'
 } uxg_fontinfo_t;
   ^
Using library U8glib at version 1.19.1 in folder: /home/bruce/Arduino/libraries/U8glib 
```

### Steps to Reproduce

1. Load Marlin code in Arduino IDE
2. Click on Verify
3. Code does not compile

**Expected behavior:** [What you expect to happen]

Code should compile normally

**Actual behavior:** [What actually happens]

Error during compilation listed above.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.
[configfiles.zip](https://github.com/MarlinFirmware/Marlin/files/2568364/configfiles.zip)

salfter, 2018-11-14:
I'm running into the same issue building under PlatformIO...but only for an Arduino Due to be installed in a Hypercube 300.  The same code builds without issue for an Arduino Mega in an Anet A8.  Configs for both are attached.

[config-anet-a8.zip](https://github.com/MarlinFirmware/Marlin/files/2583062/config-anet-a8.zip)
[config-hc300.zip](https://github.com/MarlinFirmware/Marlin/files/2583063/config-hc300.zip)

```
In file included from Marlin/src/lcd/lcdprint.h:16:0,
from Marlin/src/lcd/lcdprint_u8g.cpp:24:
Marlin/src/lcd/dogm/u8g_fontutf8.h:19:16: error: redefinition of 'struct _uxg_fontinfo_t'
typedef struct _uxg_fontinfo_t {
^
In file included from Marlin/src/lcd/lcdprint_u8g.cpp:23:0:
Marlin/src/lcd/u8g_fontutf8.h:20:16: error: previous definition of 'struct _uxg_fontinfo_t'
typedef struct _uxg_fontinfo_t {
^
In file included from Marlin/src/lcd/lcdprint.h:16:0,
from Marlin/src/lcd/lcdprint_u8g.cpp:24:
Marlin/src/lcd/dogm/u8g_fontutf8.h:25:17: error: invalid type in declaration before ';' token
} uxg_fontinfo_t;
^
Marlin/src/lcd/dogm/u8g_fontutf8.h:25:17: error: conflicting declaration 'typedef int uxg_fontinfo_t'
In file included from Marlin/src/lcd/lcdprint_u8g.cpp:23:0:
Marlin/src/lcd/u8g_fontutf8.h:26:3: error: 'uxg_fontinfo_t' has a previous declaration as 'typedef struct _uxg_fontinfo_t uxg_fontinfo_t'
} uxg_fontinfo_t;
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs/DUE/src/src/lcd/lcdprint_u8g.cpp.o] Error 1
```

Both are configured with the same type of display (Reprapdiscount Full Graphic Smart Controller).  The A8 uses a RAMPS 1.4, while the Hypercube uses a RAMPS-FD 2.x.

boelle, 2019-02-21:
@brucehvn problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead