MarlinFirmware/Marlin #12391: [bugfix 2.0.x] Test Issue on STM32F1

Weruminger, 2018-11-10:
### Description

Found out on PR#12389 that an error recognized on all other platforms hven't been recognized on STM32F1. Because the the issue was related to M106/M107 change I cannot imaging why it is not recognized on that plattform. 

Because the other platform related tests ran in failed state, there is no high pressure to have a look on it, but if there is work on platform related tests thi issue shall be investigate and solved.

I can not see that the root cause of the fails on other platforms will not be recognized on STM32F1.

### Steps to Reproduce

 have a look on
https://travis-ci.org/MarlinFirmware/Marlin/builds/453248476

**Expected behavior:** All platform related test shall failed.

**Actual behavior:** All platform related test except the STM32F1 failed.

#### Additional Information

(https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Weruminger, 2018-11-11:
It is not the code to be tested i pointed on. it was the test it self. of course i fixed the code related to the failed test results. And so newer builds succeeded. 

But on the buggy code i did expect, that all platform tests failed because it is not a platform specific code part. 

If it help, i can revert the code to the buggy version to show the behaviour,

Ludy87, 2018-11-11:
not necessary did not understand it correctly

AnHardt, 2018-11-11:
We have seen that before when the MIXER test in STM32F1 did not bark about the overflow. 

Something is wrong there.

p3p, 2018-11-11:
The same feature tests are not ran on all platforms as not all features are supported, the AVR tests are designed to have the most code coverage on the platform independent code paths, while the other platforms tests generally cover the platforms HAL implementations, but saying that I'm not sure how the STM build didn't have that error it seems generic ..