MarlinFirmware/Marlin #12392: [bugfix 2.0.x] Incorrect Z position after Auto Bed Leveling but home is ok

albertolg, 2018-11-10:
AUTO_BED_LEVELING_LINEAR enabled

When i issue G28 all is ok. Test with paper passed with some friction as i expected by my calibration.
after G29 the nozzle is higher than it should be, seems like it is about 1mm. higher

Step to reproduce:
-Home (from display or G28)
-Move Z to 0
-put a paper between nozzle and bed (position is ok)
-Auto bed leveling (from display or G29)
-Move Z to 0
-now the height is too much

P.S.
the height of right of my bed is 1mm. less than the left

MagoKimbra, 2018-11-10:
Are you sure of the value of Z_PROBE_OFFSET_FROM_EXTRUDER -0.74 because this value is used to calculate Z.

gloomyandy, 2018-11-10:
Is your bed actually flat (but with a tilt)? LINEAR ABL is really only suitable for beds that are pretty flat. If your bed has dips in it (and most do), then you really need to use a more sophisticated form of levelling (like bilinear or UBL). Linear basically just fits a plane to the three sense points, this can easily result in the height at 0,0 being wrong (unless 0,0 is one of your probe points).

Oh and as above you really have to make sure that your probe offsets are set correctly especially Z but errors in X and Y can also create some strange results.

albertolg, 2018-11-11:
I calculated the Z offset following Thomas Sanlanderer's guide https://www.youtube.com/watch?v=EcGFLwj0pnA

if I change to AUTO_BED_LEVELING_BILINEAR and increase the GRID_MAX_POINTS_X to 3 now the offset seems right

albertolg, 2018-11-11:
> Is your bed actually flat (but with a tilt)? LINEAR ABL is really only suitable for beds that are pretty flat. If your bed has dips in it (and most do), then you really need to use a more [...]

how can I be sure my bed doesn't have dips? can you tell me an easy example to check it? thanks

albertolg, 2018-11-11:
just checked the bed, is only tilted on a side, but seems really strange that I have this results with the AUTO_BED_LEVELING_LINEAR, and with AUTO_BED_LEVELING_BILINEAR I have to rise the first layer to more than 150% of the value or the first layer will be too much pressed

gloomyandy, 2018-11-11:
There are almost certainly better places than this to get general 3D printer support. You might want to try and find a forum that covers your particular printer, or you can try the Marlin facebook group or discord group. These threads are really intended for actual bugs with the Marlin firmware.

albertolg, 2018-11-11:
I'm not asking a support, just reporting my experience, if is not needed then I will just close this issue so you can go ahead with your fantastic bug-free firmware, regards

closed 2018-11-11 by albertolg

InsanityAutomation, 2018-11-11:
I don't think anyone is calling it a bug free firmware, the point they're trying to make is to go the support route first and confirm that it is not a configuration or mechanical setup issue, which seems likely from what's been said so far, and then come back and file a bug report with enough information to show it's actually a firmware bug. The fact is from what you've been through and tried so far it's too early to say firmware issue and therefore this is not the right channel for it.

Roxy-3D, 2018-11-12:
> @InsanityAutomation <https://github.com/InsanityAutomation> I'm not
> pissing on the marlin, you guys did a great job and I thank you for all the
> work you put in this. I did any kind of test and I can confirm that tilting
> the bed the auto-level feature is not working, unless I misconfigured
> something, and this is why I attached my configuration file. I followed the
> rules to report an issue, I also found some issues regarding my same
> problem #4612 <https://github.com/MarlinFirmware/Marlin/issues/4612>
> #10907 <https://github.com/MarlinFirmware/Marlin/issues/10907> so I'm not
> alone, I can also make a video of this problem, but, if you think this is
> as it should work and I have a problem because my bed is tilted then what
> is the purpose of the ABL?

> how can I be sure my bed doesn't have dips? can you tell me an easy
example to check it? thanks

A mesh size of 3 x3 is questionable.   Bump it up to 10 x 10.   And there
are instructions on the forum and the source code to dump the mesh in the
form you can plot the contour of your print bed surface in Excel.   You
will be able to see the shape of your print surface.

thinkyhead, 2018-11-12:
Be sure to probe with the nozzle and bed heated, because these expand when heated. Use `BABYSTEPPING` and `BABYSTEP_Z_OFFSET` when starting your next print. And use `M500` to save the result. This will help ensure a more accurate Z probe Z offset.

gloomyandy, 2018-11-12:
@albertolg to answer your question about should the nozzle be the same height above the bed after a G28 and then after enabling bed levelling, then the answer is "it depends". Basically if your home probe point (note this is not the same as your actual home point due to the offset of the probe) for Z is not in exactly the same location as one of the probe points used for bed levelling then it is likely that the Z height for home and Z height after levelling will be different. Basically  after bed levelling the Z height at the home position when at Z=0 will be on a line between the two nearest probe points if say the bed dips at your home point then these two probe points may be higher then your home point and so the position of the nozzle will be higher (Imagine stretching a wire across a valley). Similarly if one side of you bed is high and the other low and you only have a small number of probe points then again after levelling the 0 point will be on a line between those high and low points (and so could easily be higher or lower then the home point).