MarlinFirmware/Marlin #12397: [bugfix 2.0.x] [Q] G34 and G29, what order?

boelle, 2018-11-11:
Can G34 and G29 be used at the same time?

I start my prints with G34 to get X axis level with bed

That ends with homing of all axis or so it looks like

Then i use G29 to probe the bed using the bilinear method at 9 points

My big doubts are that the G29  cancels the work G34 has done, so is there a correct order to make use of both G34 and G29 ?

RudyBenner, 2018-11-11:
G28, then G34, then with bed heated, G29 as follows:

M190 S60 ; bed heat on
G28 ;Home
G29
M190 S0 ; heat off
G28 Z
M420 P1 Z2
M500

boelle, 2018-11-11:
why G29?

thinkyhead, 2018-11-11:
Your question pertains to which order for `G28`, `G29`, `G34`. It is easily assumed you want to do all three: home, align the ends, and level the bed.

boelle, 2018-11-11:
yes, but it seems that after G34 i homes again on its own and that then the Z align goes away

i forgot to say that i home to Z max (dual z stepper and dual z endstops)

boelle, 2018-11-11:
my start G code:

```
G28 Z; Home Z
G28 Y; Home Y
G28 X; Home X
G34
G29
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G92 E0 ; zero the extruded length
G1 X50 Y0 F3000 ; go to the middle of the front
G1 Z0.1 ; close to the bed
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 X100 E10 F200 ; make a thick line to prime extruder
G92 E0 ; zero the extruded length
G1 Z0.3 ; lift Z
G4 S2   ;wait 2 sec to bleed off excess pressure
```

i do homing of the axis in order so that wires to x carriage dont get pinched

boelle, 2018-11-11:
and i do the probing cold so that heat does not influence the pinda probe, of course my probe offset is set so that the first layer is perfect.

boelle, 2018-11-11:
maybe i'm missing the M420 P1 Z2
and M500

TheLongAndOnly, 2018-11-13:
So you need to home before doing z-alignment. At the end of G34 the system homes again which should keep the result of G34 but compensate loss of the absolute z-position. Afterwards you can apply the UBL functions.

boelle, 2018-11-13:
that suggest my start code is correct, one could ask why i do the G34 at every print but i thought that if a print sticks well to the PEI i could risk getting it a bit out of alignment.

and then again i might think the wrong way

maybe its bettre to do a G34 only every now and then and save it to eeprom and just do the G29 at every print

boelle, 2018-11-14:
G29 ?? did you mean G24?

thinkyhead, 2018-11-14:
No, I meant `G29` — which is why I referred also to `M420`.

boelle, 2018-11-14:
G29 is the 9 point probing, and if a print sticks very well it could be that i get it out of what when removing the print, that is why i do it

and my question is still if a G34 followed by G29 cancels out the G34 ? and if the better option would be to do the G34 only every now and then and save that to EEProm and call it from start code by (i dont know how to save G34 data as i forgot and there is no doc on http://marlinfw.org/ yet)

TheLongAndOnly, 2018-11-14:
G34 ja a pure mechanical alignment of the different steppers driving the z-axis. No need to store anything after the event. This should also explain that it is fully independent of G29. Nevertheless I would recommend to do G29 always after G34 as likely results will be suboptimal if you do it the other way around.

closed 2018-12-06 by boelle