MarlinFirmware/Marlin #12399: compile error

hamlet600, 2018-11-11:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.
# @ @

hamlet600, 2018-11-11:
i use bugfix2 when uncomment           #define BABYSTEP_ZPROBE_OFFSET     // Combine M851 Z and Babystepping

error in compile   sketch\src\lcd\menu\menu.cpp: In function 'void lcd_babystep_zoffset()':

sketch\src\lcd\menu\menu.cpp:374:9: error: 'thermalManager' was not declared in this scope

         thermalManager.babystep_axis(Z_AXIS, babystep_increment);

please guide me

gloomyandy, 2018-11-11:
Please supply your configuration files. How recent is your bugfix-2.0.x? Please download the latest version and confirm that the problem still exists.

Ludy87, 2018-11-11:
@gloomyandy ``THERMAL_PROTECTION_HOTENDS`` and/or ``THERMAL_PROTECTION_HOTENDS`` been disabled 

https://github.com/MarlinFirmware/Marlin/blob/a0c795b097a30eff006c8dff178abf5f1f1907fa/Marlin/src/lcd/menu/menu.cpp#L39-L41

AnHardt, 2018-11-11:
Try 
At the top of `\src\lcd\menu\menu.cpp`
```diff
@@ -33,11 +33,11 @@
 
 #if ENABLED(EEPROM_SETTINGS)
   #include "../../module/configuration_store.h"
 #endif
 
-#if WATCH_HOTENDS || WATCH_THE_BED
+#if WATCH_HOTENDS || WATCH_THE_BED || ENABLED(BABYSTEP_ZPROBE_OFFSET)
   #include "../../module/temperature.h"
 #endif
 
 #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
   #include "../../module/probe.h"
```

hamlet600, 2018-11-12:
it is last version bugfix2 and thermal protection for both hotend and bed are disabled

hamlet600, 2018-11-12:
> Try
> At the top of `\src\lcd\menu\menu.cpp`
>
> ```diff
> @@ -33,11 +33,11 @@
>
>  #if ENABLED(EEPROM_SETTINGS)
>    #include "../../module/configuration_store.h"
>  #endif
>
> -#if WATCH_HOTENDS || WATCH_THE_BED [...]

it's work thank you.

commit be627176 (2018-11-12): Fix BABYSTEP_ZPROBE_OFFSET (#12408)

Fix #12399

Did not compile without  `WATCH_HOTENDS || WATCH_THE_BED`.
Add condition to include "../../module/temperature.h".
closed 2018-11-12 by hamlet600