MarlinFirmware/Marlin #12400: bed level

Linderkin, 2018-11-11:
> You should also do a Motion > Bed Leveling > Level bed followed by Store Settings to ensure a perfect leveling.

after the calibration, it is put in Stow Z-Probe mode and if you give it to accept, it drops and drags the sensor by the bed

Anycubic Kossel Linear Plus

thinkyhead, 2018-11-13:
You have not provided enough information.

Linderkin, 2018-11-14:
Marlin 2.0 Following the steps of the readme

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/config/examples/delta/Anycubic/Kossel

Steep --> You should also do a Motion > Bed Leveling > Level bed followed by Store Settings to ensure a perfect leveling.

thinkyhead, 2018-11-14:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2018-11-14 by thinkyhead