MarlinFirmware/Marlin #12403: Layer shifts on BugFix-2.0.0

Roxy-3D, 2018-11-12:
Is anybody else seeing layer shifts on BugFix 2.0.0 ?   I'm seeing one or two of them on long prints.   I'm seeing them on the X-Axis (and mostly in the negative X direction).

I'm running an IDEX machine.    So mostly, I'm asking if anybody is seeing layer shifts on non-IDEX machines.

Anybody???

InsanityAutomation, 2018-11-12:
I'll chime in quickly to confirm multiple idex machines, and rolling back to 1.1.9 problem vanishes. Working on matching settings and trying to reproduce elsewhere now.

nemphys, 2018-11-14:
If you have S-Curve acceleration, try disabling it and check if you still get layer shifting (see https://github.com/MarlinFirmware/Marlin/issues/12398).

thinkyhead, 2018-11-14:
If possible, please narrow down to a specific date where the problem starts to manifest. This can be done in LOG2(n) steps by following this procedure:

- Test October 1 and see if the problem exists.
  - Yes? << Test September 1 and see if the problem exists.
    - No? >> Test September 15 and see if the problem exists.
      - Yes? << Test September 8 and see if the problem exists.
        - Yes? << Test September 4 and see if the problem exists.
          - No? >> Test September 6 and see if the problem exists.

In this manner the issue can be narrowed down to a specific date.

Christophe76350, 2018-11-14:
Hello,

I have the same problem on a CoreXY with a MKS SBase motherboard.

I have done several tests in my case the deactivation of the S-curve acceleration does not change anything.

I will test with the code versions as you request tonight.

After having made many attempts to print the same G-code, the shift of the layers is not random. All the parts I have printed that all have the same offset (all the parts are identical).

thinkyhead, 2018-11-14:
That is helpful information. I have so far not been able to reproduce the issue, so if it can be narrowed down to some specific change that will be very helpful.

InsanityAutomation, 2018-11-14:
To keep some of the discord info here as well, first user reports of shifts came to me on October 20th. I'll have another machine here Friday to give me more opportunity to test, since it only occurs on long prints it'll be a bit slow going to track down.

Roxy-3D, 2018-11-14:
> To keep some of the discord info here as well, first user reports of shifts came to me on October 20th.

My experience is the same.   Using a branch from October 20th...   I am *NOT* seeing the failure.

Christophe76350, 2018-11-15:
Test with the version of October 29th last night same problem. I do a test with the version of October 15th tonight.

InsanityAutomation, 2018-11-15:
I flashed Oct 13th last night and am running things now.

Christophe76350, 2018-11-15:
version of October 15th same problem for me.

Christophe76350, 2018-11-15:
October 1st version no layer shift for me it's okay.

Roxy-3D, 2018-11-15:
> October 1st version no layer shift for me it's okay.

I'm not sure that is true!   Would you mind doing another long print with it please?   I'm pretty sure I saw a dual X & Y layer shift with the September 21st code.

Christophe76350, 2018-11-16:
First photo firmware on October 1.
Second photo firmware dated October 15th.

The screws show the offset of the layers well.

[image]
[image]

Atatoth, 2018-11-16:
Omg so much??? Good to know I have 1.1.9 bugfix for now

thinkyhead, 2018-11-16:
So, sounds like some change between Oct 1-15 is affecting some (so next try Oct 8). And maybe something from around Sept 21 is affecting others (so next try Sept 25). I'll look at changes between Oct 1-15 and see if there are any suspect commits.

italocjs, 2018-11-16:
i'm losing steps with the latest 2.0.x too, i'll try to rollback to 1.1.6, which some people are reporting is working fine.   I'm using a corexy with tmc2130

Roxy-3D, 2018-11-16:
What low level things are different between 2.0.0 and 1.1.6 ?     It is probably not a stepper driver losing steps because of timing differences.   The reason I say that is with IDEX machines in Duplication mode, both extruders are losing position at the same time (and the same amount).

It feels more like a set_directions() call is missing or getting out of phase with other things happening...

InsanityAutomation, 2018-11-16:
I still had issues with Oct 13th here. Just flashed Sept 27th and I'm running again. I'm using the same 30hr gcode for every test. 2nd idex machine got delayed, didn't see it occur on the singlenozzle crx. Still more to go...

Roxy-3D, 2018-11-17:
> I still had issues with Oct 13th here. Just flashed Sept 27th and I'm running again. I'm using the same 30hr gcode for every test. 

I'm using the exact same .GCode file.   And on that print, I had a small X Axis shift after about 10 hours.     And that is with the Sept. 21st code base (bugfix 2.0.0).

thinkyhead, 2018-11-17:
If Sept 21 is bad, then try Sept 1 and see if it works. If it does, then try Sept 10, and so on… Too bad we can’t find a short print that consistently causes the issue to occur.

I will look at the `set_direction` behavior and see if anything obvious stands out.

InsanityAutomation, 2018-11-17:
I'm going all the way to the 1.1.9 release date of 2.0 since we don't see it in 1.1.9 but have yet to find a good 2.0 date. Just built aug3rd (3 days forward but the couple commits are minor and stop idex crashing was pulled from 1.1)

Roxy-3D, 2018-11-18:
> If Sept 21 is bad, then try Sept 1 and see if it works. 

Actually....  I've gone a different path.   I loaded up the current 1.1.9 and back ported a few things I need into it.    I'm 8 hours into a print and so far, no problems...    If the print finishes OK, I'll kick off another large print on it to confirm the problem is not on 1.1.9.

thinkyhead, 2018-11-18:
I wish that would tell us about 2.0, but who knows where the problem lies? It might be associated with display update, serial communication, G-code parsing, or any number of things. But if you happen to port something into 1.1.9 that causes bad behavior, that would be a great find.

InsanityAutomation, 2018-11-18:
Fyi I'm using SD to eliminate serial errors and monitor from my PC while it's running.

italocjs, 2018-11-18:
@InsanityAutomation  my screen and sd support are disabled, i'm printing trough octoprint.  so i dont believe they are causing the problem.   i

Roxy-3D, 2018-11-18:
> I wish that would tell us about 2.0, but who knows where the problem lies? 

Mostly...  I'm looking for a point in time without the problem.   (and I do have some large prints I have to get done)     But I'm starting to suspect the planner:: changes to merge lines  to avoid stutter.

It would be good to have a 2.0.0 version that had planner:: stutter changes taken out of it....   I think there is a 50%-50% chance the problem goes away if we do that.

Christophe76350, 2018-11-18:
After a lot of testing since my last message my layer shifts occur if I enable "Z_STEPPER_AUTO_ALIGN" during compilation.

Whether I use G34 or not in my G-code I have layer shifts.

If I compile the firmawre without the "Z_STEPPER_AUTO_ALIGN" my prints are correct without layer shift.

Christophe76350, 2018-11-18:
That's why I think my problem is not necessarily the same as the one being sought here.

In all this in my case it has an influence on my layer shift. But the tests I did on different versions of the firmware certainly don't give any indication of your problems.

One thing is certain after having performed a tenth test for me the activation of Z_STEPPER_AUTO_ALIGN causes a layer shift for sure.

I'm not saying that this is not a different problem I'm just giving the results of my tests.

Roxy-3D, 2018-11-18:
> One thing is certain after having performed a tenth test for me the activation of Z_STEPPER_AUTO_ALIGN causes a layer shift for sure.

Thank You for the careful analysis!!!

Roxy-3D, 2018-11-18:
In your prints....   Do you see the failure at the same place in the print?  Or does it fail at random times?   Int the layer shifting I'm seeing, it happens at random times.   

But I'm very sure it is not just a stepper motor that wasn't strong enough be cause some times it has both X & Y shifted.    And @InsanityAutomation  has seen the layer shift on an IDEX machine in Duplication mode.   So it wasn't just one stepper that lost steps.  It was two at the same time.

thinkyhead, 2018-11-18:
Confirmed that it was the X axis shifting in duplication mode? Of course Y shift in duplication mode would have no special significance.

Roxy-3D, 2018-11-18:
> And @InsanityAutomation has seen the layer shift on an IDEX machine in Duplication mode. So it wasn't just one stepper that lost steps. It was two at the same time.

> Confirmed that it was the X axis shifting in duplication mode? Of course Y shift in duplication mode would have no special significance.

That is what I remember.   I'm sure he will jump in with confirmation soon.  (At the time we were talking about multiple steppers losing position at the same time.  That is why I think this is true.)

InsanityAutomation, 2018-11-18:
Yes, it was x that shifted both parts in duplicate mode. I'll dig up the photos from the end user who first reported it later tonight.

kAdonis, 2018-11-18:
@Christophe76350 
you could try to increase stepper pulse width manually in Configuration_adv.h :

`#define MINIMUM_STEPPER_PULSE 4`

I had layer shifts with CoreXY, LPC1768 and DRV8825 that lead to "skewed" prints like in your picture
Issue #11047

Christophe76350, 2018-11-19:
> In your prints.... Do you see the failure at the same place in the print? Or does it fail at random times? In the layer shifting I'm seeing, it happens at random times.

The layer shifts are all the same with each impression. There is nothing random, I had a moment to question the mechanics but no problem on this side. I have a second mechanically identical one that works with a Rhumba motherboard and the no layer shift on the same Gcode with the firmware compiled with Z_STEPPER_AUTO_ALIGN enabled.

> @Christophe76350
> you could try to increase stepper pulse width manually in Configuration_adv.h :
> 
> `#define MINIMUM_STEPPER_PULSE 4`
> 
> I had layer shifts with CoreXY, LPC1768 and DRV8825 that lead to "skewed" prints like in your picture
> Issue #11047

I'm currently testing.

xenovacivus, 2018-11-20:
Issue #12491 could be related, but would mean this layer shifting is due to skipped steps.

A couple ideas:
1. Test with unloaded steppers and see if they maintain position
2. Slow down the stepper ISR (to ~1/4th the frequency).  Keeping all curve computation the same while slowing down the step rate would reduce the chance for missed steps.  But if it's a logic issue, the skew will still be apparent (unless it's a timing issue/interaction with another ISR).

thinkyhead, 2018-11-20:
Don't give up trying to narrow down the specific day that the issue appeared. These threads tend to go on forever with speculations, when the best use of time would be to meticulously seek out the specific commit that introduced the problem.

InsanityAutomation, 2018-11-20:
I'm 30hrs in on the Aug 3rd snapshot with no issues. From there we can safely say the issue appeared in 2.0 later, and was not a result of the refactoring. I have another machine now validating Sept21st however if it's anything like the bugs fixed for 1.1.9 some machines never saw it and some couldn't get rid of it.... I honestly don't know how to ensure I hit the worst case scenario here to force a machine to skip steps, so if the machine has never skipped I don't know if it ever will....

landodragon141, 2018-11-20:
Have we checked for a correlation between stepper driver types or motherboard?

InsanityAutomation, 2018-11-20:
Fyi, brand new machine out of the box with factory firmware dated sept21st layer shifts on X using the same gcode I'm running on the other machine. Aug 3rd snapshot now has 38hrs with no shift.

InsanityAutomation, 2018-11-21:
Worth noting one machine is all stand-alone tmc2208 spreadcyle, the other is 4988 on both X and tmc standalone spreadcyle on y.

Roxy-3D, 2018-11-21:
>  I have another machine now validating Sept21st however if it's anything like the bugs fixed for 1.1.9 some machines never saw it and some couldn't get rid of it.... 

I have also seen the layer shifts with a September 21st code base.    Right now, I think we have to book ends.    The problem seems to have been introduced somewhere between **August 3rd and September 21st.**

NKote, 2018-11-21:
I'm also have this nasty issue. D-bot CoreXY 1.7A steppers.
Few days ago changed RAMPS 1.4+mega2560  w marlin 1.1.8 for RURAMPS4D 1.3 + DUE.
TMC2130 SPI on X,Y,E . TMC2208 in standalone on Z. Homing with endstops. TMC2130 lib v 2.5.0. 800ma/16 steps in adv.cfg. Tried to set from 700ma to 1000ma without any result, switcing on/off  hybrid mode didb't help as well. What left ? Try to disable stealthchop, S-curve, older TMC 2130 lib ?
Shifs occur at speeds over 50mm/s. Acceleration is set to 1000.
I'm kinda tired, almost every print is ruined. Wanted to increase printspeed, but did increase a hedache.

ErikFontanel, 2018-11-21:
Not sure if my issues are related, but I experience extreme shifting with TMC2208:

[image]

Although I'm still dialling in the new configuration, this x-axis shift happened a few times already with the same print. It's a fairly simple bracket printed through octoprint. Est. print time  was 1:35 and the . shift started right away but became really noticable around 5%. The x-axis eventually is so far off, the hot-end keeps banging on the x-min stop… This happened with manual mesh bed levelling, tonight I'll try again with turned levelling off. No problems at all with homing or manual movement via Gcode, printing via Cura through USB or Octoprint or setting a higher/lower current.

Machine: Ender-3
Board: [Fysetc F6 v1.3](https://wiki.fysetc.com/F6_V1.3/)
Drivers: TMC2208 drivers (Watterott, configured through software serial)
Power: 24v
X Stepper current: 700mA
Microsteps: 16
Mode: Stealthchop (hybrid mode off)
Marlin: 0946cbcdca99d833d1a4667d00daf112394107c0 (but this happened with older commits too)

NKote, 2018-11-21:
I've tryed some print and found that S-Curve acceleration and different TMC2130 libraries doesn't affect the shifting at all. decreasing speed to 50mm\s or switching to spreadcycle (hell, its noisy!!!) helps. And the drivers are really hot. 60 degrees at least, although they are fan-cooled. With RAMPS 1.4. it was  OK even at 80mm\s.

thinkyhead, 2018-11-22:
Looks like we're getting closer. Let's try Sept 1st next!

InsanityAutomation, 2018-11-22:
I have a print 25hrs in on a sept1st snapshot. No shifts yet.

italocjs, 2018-11-24:
Question not really related: i reverted back to the 1.1.9 marlin, and i'm setting things up. Which hybrid threshold is a good start value for a 1.7a 4kg stepper?  i heard is best to keep spreadcycle for high speeds

Roxy-3D, 2018-11-24:
> I'm kinda tired, almost every print is ruined. Wanted to increase printspeed, but did increase a hedache.  Could anybody post a link to Pre-August 2.0 build ?

> Question not really related: i reverted back to the 1.1.9 marlin, and i'm setting things up.

I understand the frustration!   But the early information points to September 1st being OK.   Instead of scampering off to all kinds of different points in the code base, it would be very helpful if the people that need a solid base all re-group at BugFix 2.0.0 as of September 1st.    That should give us several things.   Most importantly,  we will get more confirmation the problem wasn't in the code base at that point of time.  

Here is a link to October 31st.   (A day earlier than the 'known working' branch just because I don't know what time of day people's snap shots of September 1st were taken.)    Please use this code base:   https://github.com/MarlinFirmware/Marlin/tree/2cff0d1f39061e2309510da0dbc8b21975f965c6

italocjs, 2018-11-25:
> I'm also have this nasty issue. D-bot CoreXY 1.7A steppers.
> Few days ago changed RAMPS 1.4+mega2560 w marlin 1.1.8 for RURAMPS4D 1.3 + DUE.
> TMC2130 SPI on X,Y,E . TMC2208 in standalone on Z. Homing with endstops. TMC2130 lib v 2.5.0. 800ma/16 steps in adv.cfg. Tried to set from 700ma to 1000ma without any result, switcing on/off hybrid mode didb't help as well. What left ? Try to disable stealthchop, S-curve, older TMC 2130 lib ?
> Shifs occur at speeds over 50mm/s. Acceleration is set to 1000.
> I'm kinda tired, almost every print is ruined. Wanted to increase printspeed, but did increase a hedache.
> Could anybody post a link to Pre-August 2.0 build ?

try using the 1.1.9 if possible, after changing to it, i'm not having any shifts, printing now at 80mm/s with mov speed of 100mm/s.  using tmc2208 with full stealthchop

kwalters3, 2018-11-27:
Using a July download of the 2.0 firmware, changed from repetier.  Has layer walking in the X and Y  when printing with OctoPrint, same Gcode prints fine when printing from SD card.  Just downloaded and configured a Nov version, downloaded 11-23.  Still have the layer walking with OctoPrint

Hardware:
Due w/ RADDS shield, DRV8825
CoreXY

Roxy-3D, 2018-11-28:
A possibility???   https://github.com/MarlinFirmware/Marlin/issues/12540#issuecomment-442267831

InsanityAutomation, 2018-11-28:
Given that the torture gcode were using is a massive amount of short jerky moves, it very much sounds like we're dealing with the same issue!

thinkyhead, 2018-11-28:
> A possibility?

Strangeness in the acceleration handling could certainly be involved. I’ll look at it and review changes made in that region of the code over the past few months.

InsanityAutomation, 2018-12-02:
Fyi, current code base with junction deviation was notably slower than with traditional jerk, after a 47hr print finished (that took 36hrs before) it only had a single shift that was half the size of what they were previously. This was between 40 and 45hrs in. Again same gcode. I should start selling these lithos with a single shift here or there just to get material cost back lol

gururise, 2018-12-03:
> That's why I think my problem is not necessarily the same as the one being sought here.
>
> In all this in my case it has an influence on my layer shift. But the tests I did on different versions of [...]

@Christophe76350 Hey, I hope you keep testing. You have definitely found something and I can absolutely confirm that #11302 is also causing significant layer shifting for me.  Been trying everything to pinpoint the cause until I saw your post and decided to comment out Z_STEPPER_AUTO_ALIGN. 

Here are before and after images of some prints of the 20mm hollow calibration cube.  The image on the left is w/o s-curve acceleration and adaptive step smoothing enabled, the right image is with those two featured turned on.

**With Z_STEPPER_AUTO_ALIGN enabled:**
[image]

**With Z_STEPPER_AUTO_ALIGN commented out:**
[image]

As you can see, with Z_STEPPER_AUTO_ALIGN commented out, there is a significant reduction in layer shifting (doesnt seem to be completely gone).

This was printed on a CoreXY machine running Nov 30 git of bugfix-2.0.0 on an LPC1768 (mks-sbase).

Roxy-3D, 2018-12-05:
I'm wondering if we should back out all planner and stepper motor changes going back to Sept. 1st.   And then slowly, and carefully start adding things back in.     And maybe not in the original merge order.   

A strategy like that might help us figure out what is causing the trouble.

InsanityAutomation, 2018-12-05:
There do appear to be quite a few combination of things having an impact here which is what's making it so difficult to pin down...

gururise, 2018-12-05:
I believe the G34 issue I posted above is related to #11205

Bob-the-Kuhn, 2018-12-05:
I had a couple of layer shifts yesterday and today.  Both were in the Y direction.

[image]

That one shifted 2.0 mm.

The next one shifted 19.2mm and later 22.9 mm.

[image]

I home to the Y MAX endstop so I deliberately ran the head into the Y MAX endstop to see what position it reported.  The endstop message said 477.72.  It should have been 436.0 so the Y positioning was off by 41.7 mm.  The two shifts totalled 42.1 mm so we have a reasonable agreement in offsets.

After hitting the endstop I immediately did a M114 and it reported a Y of 600 with a Y count of 189666.

Next I did a G28 which gave me a Y of 436 with a count of 173102.  

The difference between the two counts divided by my steps-per-mm give a distance of 41.72 mm.

Something definitely messed with the coordinate system.

The only recent changes I've made were to enable JUNCTION_DEVIATION , ADAPTIVE_STEP_SMOOTHING and  LIN_ADVANCE.  I do NOT have S curve acceleration nor Z_STEPPER_AUTO_ALIGN enabled.

I'll back those three changes out and restart the last print.  The first shift happened at the 10-12 hour point so it'll be late Wednesday before I have confidence that the shift has gone away.

[420 more replies]

closed 2020-07-06 by ellensp