MarlinFirmware/Marlin #12405: LPC1769: Soemething wrong with usbserial?

alexxy, 2018-11-12:
Seems like serial somehow messed up.. (timeouts.. etc..)

```
ttyACM1 70°> m502
SENDING:M502
Hardcoded Default Settings Loaded
ttyACM1 70°> m503
SENDING:M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75     
M200 T1 D1.75  
per unit:      
M92 X107.06 Y107.06 Z1920.00 E300.00
Maximum feedrates (units/s):
M203 X100.00 Y100.00 Z10.00 E10.00
Maximum Acceleration (units/s2):
M201 X1020000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X0.00 Y0.00 Z0.000
Unif           
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
M145 S1 H240 B110 F0
PID settings:  
M301 P13.99 I0.77 D63.74
M304 P291.47 I38.78 D547.64
LCD Contrast:  
M250 C140      
Z-Probe Offset (mm): (mm):
M851 Z-1.50    
ec             
ttyACM1 70°> 
ttyACM1 70°> 
ttyACM1 70°> m503
SENDING:M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75     
M200 T1 D1.75  
M200 D0        
Steps per unit:
M92 X107.06 Y107.06 Z1920.00 E300.00
Maximum feedrates (units/s):
M203 X100.00 Y100.00 Z10.00 E10.00
Maximum Acceleration (units/s2):
ech            
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
M145 S1 H240 B110 F0
PI0.77 D63.74  
M304 P291.47 I38.78 D547.64
LCD Contrast:  
M250 C140      
Z-Probe Offset (mm): (mm):
M851 Z-1.50    
Stepper driver current:
ttyACM1 70°> 
ttyACM1 70°> 
ttyACM1 70°> m503
SENDING:M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75     
M200 T1 D1.75  
M200 D0        
Steps per unit:
M92 X107.06 Y107.06 Z1920.00 E300.00
Maximum feedrates (units/s):
M203 X100.00 Y100.00 Z10.00 E10.00
Maximum Acceleration (units/s2):
M201 X10rk> E<max_e_jerk>
M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X0.00 Y0.00 Z0.000
Un             
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
M145 S1 H240 B110 F0
PID settings:  
M301 P13.99 I0.77 D63.74
M304 P291.47 I38.78 D547.64
LCD Contrast:  
M250 C140      
Z-Probe Offset (mm): (mm):
M851 Z-1.50    
ec             
ttyACM1 70°> 
ttyACM1 70°> 
ttyACM1 70°> 
ttyACM1 70°> m503
SENDING:M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75     
M200 T1 D1.75  
300.00         
Maximum feedrates (units/s):
M203 X100.00 Y100.00 Z10.00 E10.00
Maximum Acceleration (units/s2):
M201 X1000.00 Y1000.00 Z20.00 E500.00
Acceleration (units/s2): P<print_ac10.00 Z0.30 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X0.00 Y0.00 Z0.000
Unified Bed Leveling:
M420 S0 Z0.00  
Unified Bed L  
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
M145 S1 H240 B110 F0
PID settings:  
M301 P13.99 I0.77 D63.74
M304 P291.47 I38.78 D547.64
LCD Contrast:  
M250 C140      
Z-Probe Offset (mm): (mm):
M851 Z-1.50    
ech            
ttyACM1 70°> 
ttyACM1 70°> 
ttyACM1 70°> 
ttyACM1 70°> m500
SENDING:M500
Settings Stored (613 bytes; crc 20015)
```

gloomyandy, 2018-11-12:
USB Serial works fine for me with LPC1768. Please provide full configuration details, details of what host program you are using, what operating system and steps on how to reproduce the problem. Also how recent a build is this? What control board are you using? Also you don't seem to be having the same issues in your other issue about UBL, so what is different between the two?

gloomyandy, 2018-11-12:
But you do not seem to be getting the same errors in the log that reports the UBL problem, so what is different? Do you see the above problems all of the time, if not when do you see them? Have you been using this setup before? Is this problem new? Has anything other than the Marlin build changed? If not what was the previous Marlin build that worked?

Do you have access to a Windows system? Do you see the same issues using that?

alexxy, 2018-11-12:
I dont have windows systems at all. And before it wasnt a problem. (I use it for ~3 years with linux and it was no problems at all). And seems latest bugfix-2.0.x produces strange behavior. Constantly printing this on console (this wasnt before)

```
busy: paused for user
Not SD printing
Not SD printing
busy: paused for user
Not SD printing
Not SD printing
busy: paused for user
Not SD printing
Not SD printing
busy: paused for user
Not SD printing
```

alexxy, 2018-11-12:
> Marlin hasn't supported the the LPC176X for three years so something has changed.
It was about using linux for 3d remote printing =)

> Have you been using the LPC176X based board before now with no problem? What was the last version of the firmware that worked? Have you made any changes to the Octoprint software? Is this the same version that has been working before?

Octoprint is 1.3.9 as i used before. Previos build (~2 weeks ago seems have no problems). I actualy use Marlin on LPC since it was supported (I added Azteeg X5 GT board support files for Marlin)

> You also didn't answer my questions about the output on your other thread. That does not seem to have any errors in the output. So what is different between that and the output you have shown in this thread?

I selected part of output that wasnt messed up. And opened this thread regarding serial output.

gloomyandy, 2018-11-12:
So how often are you seeing the dropped characters? It is clearly not all the time. Do you have any idea what triggers the loss of output? Does it do it straight away from a printer boot?

When you get chance please try going back to the older version of the firmware and confirm that there are no problems. Please post the details of the version that you tested.

I'm asking for all of this because I can't reproduce this issue at all (I've been testing USB serial a lot over the last few weeks). I've also not seen any other reports of problems from anyone else, so I don't really have much to go on. There have not been many changes to the USB serial code since p3p merged his big changes a few weeks ago, it would be useful to know if the problem was caused by those changes.

gloomyandy, 2018-11-12:
Oh and I notice that in your config files you set the baud rate to 115000 can you try running with it set to 250000 (I suspect it will not make any difference, but it is something that is different from the standard settings). Was there a reason for changing it?

p3p, 2018-11-12:
Not sure this is the problem but about 3 weeks ago we did add the code to drop characters rather than block on transmission buffer overflow in the CDCSerial, if the host is too slow to keep up this would have been hidden behind the blocking transmission, we came to the consensus that dropping characters was better than blocking but maybe your suggestion that a maximum of 2ms blocking was too short was correct.

The baud rate doesn't effect CDC Serial, it runs at usb speeds.

gloomyandy, 2018-11-12:
@p3p yes it could be the timeout. But what is really odd is the way that things like "echo" are truncated in the original post. I can understand if it is only commands that produce large amounts of output, but I'm not sure if that is always the case. I assumed that the baudrate should not make any difference but the fact that it had been changed from the default just made me wonder. Do you know what the previous problem was? Was it with serial rather than USB?

That change was in the framework, is there an easy way for @alexxy to edit that code before building?

alexxy, 2018-11-12:
If you tell me what to do i can do it.

alexxy, 2018-11-19:
Example subsequent execution of M503

```
ttyACM0 18°> m503
SENDING:M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75     
M200 T1 D1.75  
cel>           
M204 P1000.00 R500.00 T1000.00
Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X0.em v1.01 inactive.
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
M145 S1 H240 B110 F0
PI             
ttyACM0 18°> 
ttyACM0 18°> 
ttyACM0 18°> 
ttyACM0 18°> m503
SENDING:M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75     
M200 T1 D1.75  
M200 D0        
Steps per unit:
M92 X107.06 Y107.06 Z1920.00 E300.00
Maximum feedrates (units/s):
M20500.00 T1000.00
Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X0.00 Y0.00 .
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
M145 S1 H240 B110 F0
PI             
ttyACM0 18°> 
ttyACM0 18°> 
ttyACM0 18°> 
ttyACM0 18°> m503
SENDING:M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75     
M200 T1 D1.75  
cho:  M204 P1000.00 R500.00 T1000.00
Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X0.00 inactive.
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
[... 245 lines cut ...]
Tool-changing: 
Z2.00          
ttyACM0 18°> m503
SENDING:M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75     
M200 T1 D1.75  
vel_accel>     
M204 P1000.00 R500.00 T1000.00
Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X0.00 v1.01 inactive.
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
M145 S1 H240 B110 F0
ttyACM0 18°> 

```

gloomyandy, 2018-11-19:
If you let me have your current config files plus any changes you may have to the pins file I'll try building you a new firmware.bin with a different serial implementation and a longer timeout on the overflow handling.

p3p, 2018-11-19:
Sorry been distracted by other things, when I looked into this I found an issue with the pending function not always working as intended I meant to patch it along with the PWM stuff but that got delayed and I forgot about this. It should help with this issue as it will actually wait for 2ms.

alexxy, 2018-11-20:
@gloomyandy config is same as #12407 
firmware is built using platformio

gloomyandy, 2018-11-20:
Do you have any changes to the pins file?

alexxy, 2018-11-21:
@gloomyandy no pin definitions are intact (actualy they were initialy made from my config)

@p3p ok. will try it

p3p, 2018-11-26:
@alexxy did this solve the issue?

alexxy, 2018-12-01:
@p3p seems working now. So i close issue

closed 2018-12-01 by alexxy