MarlinFirmware/Marlin #12409: Marlin bugfix-2.0.x virtual extruder problem

orhaan68, 2018-11-12:
Dear friends, i just flash my 3D printer (ramps 1.4) with marlin bugfix 2.0.x . Previously i was using marlin 1.1.9 and i have problems with diamond hotend and mixing extruder. At 1.1.9 I used mixing ratio at repetier host with M163 and M164 commands. But steppers moves more than command at mixing. At bugfix 2.0.x steppers move as command at mixing just with M165 command. M163 and M164 do not works. When I select virtual extruder T0 or T1 etc, it said unknown command.
repetier host virtual extruder code;

Ext1;
M163 S0 P1.0
M163 S1 P0.0
M163 S2 P0.0
M164 S0

Ext2;
M163 S0 P0.0
M163 S1 P1.0
M163 S2 P0.0
M164 S1

Ext3;
M163 S0 P0.0
M163 S1 P0.0
M163 S2 P1.0
M164 S2
.
.
.
.
Ext15;
M163 S0 P0.33
M163 S1 P0.33
M163 S2 P0.34
M164 S15

AnHardt, 2018-11-12:
Configs please.

orhaan68, 2018-11-13:
The configuration file is attached.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2574748/Configuration.txt)

AnHardt, 2018-11-13:
Are you sure your version is newer than #12375.

AnHardt, 2018-11-13:
You might want to test #12417

thinkyhead, 2018-11-16:
I'm unable to reproduce the issue, using the current `bugfix-2.0.x`. `T0`-`T15` give no error, but `T16` is out of range.

AnHardt, 2018-11-16:
Default number of v-tools is 16 - so T0 to T15 is correct.
If all is as planned, the 'auto retraction tool', if defined, gets T16 - but is unselectable by purpose.

Probably fixed by  meged #12417.

boelle, 2019-02-21:
@orhaan68 problem solved?

orhaan68, 2019-02-22:
Yes i solve problem. Tank you for your kind helps.

Android’de Yahoo Postadan gönderildi

boelle, 2019-02-22:
@orhaan68 click green close button below

closed 2019-02-22 by orhaan68