MarlinFirmware/Marlin #12410: Bed leveling fade height issue

CageFox, 2018-11-12:
### Description

Fade height seems to be ignored or completely not working

### Steps to Reproduce

1. With bilinear leveling enabled and fade height 2mm start to print

**Expected behavior:** [What you expect to happen]

a. Active compensation of bed height problems, z steppers moves with xy moves. 
b. When printed object will be 2mm or more in height this compensation whould be stopped, no z movements with xy movements

**Actual behavior:** [What actually happens]

a is correct, bed height compensate movements presents
b is never happened, not in 2 mm height, not in ever 58mm - z still moves with xy moves

#### Additional Information
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2572098/Configuration.zip)

Start script:

G28 ; home all axes
G29 ;
M420 S1 Z2; Turn on bed leveling

GMagician, 2018-11-12:
in configuration just `#define ENABLE_LEVELING_FADE_HEIGHT` no numbers after

CageFox, 2018-11-12:
> no numbers after
I see. And how to set default fade height then?

CageFox, 2018-11-12:
I mean default value, value used without M420 Z. Is there way?

RudyBenner, 2018-11-13:
I guess one could try it, put a number there and see if it spits an error message.

InsanityAutomation, 2018-11-13:
Won't work. Only referenced one place in the code and used as a bool. Discussed and turned down adding as a default awhile ago.

CageFox, 2018-11-13:
Thanx, closing issue

closed 2018-11-13 by CageFox