MarlinFirmware/Marlin #12411: Resume not happen after pause

hamlet600, 2018-11-12:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-11-13:
How are you pausing and resuming? Host printing or SD printing?

InsanityAutomation, 2018-11-13:
Most likely octoprint. They made a change on 1.3.7 that delayed state changes so they miss the resume command. Octoprint bug, not marlin if so.

hamlet600, 2018-11-13:
i use pause from lcd(128x64) menu

e-Mole, 2018-11-14:
Same as my strange printing pause (from LCD menu) behavior:

1. head is going to park position - o.k.
2. on LCD is now message about changing filament - wrong (but message only)
3. after enter (click) - screen with continue or purge option - o.k.
4. after continue (click) - head is going to last print position and will stand - wrong (why not continue printing now?)
5. on LCD is Info screen now - o.k.
6. it is necessary to enter menu (click) and select Resume - wrong (why confirm second time?)
7. after confirmation then print continue...

thinkyhead, 2018-11-14:
> wrong (but message only)

If you're still seeing that then you should update.

thinkyhead, 2018-11-30:
This should be a bit better now. `M125` was checking for SD print, but with SD print already paused by `M25` or LCD-based pause, it was getting the wrong answer.

boelle, 2019-02-21:
@hamlet600 problem solved?

e-Mole, 2019-02-21:
@boelle In my case - I did not test the last version. In January, however, this issue was still up to date.

boelle, 2019-02-21:
we always need to test with latest bugfix 2.0

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead