MarlinFirmware/Marlin #12420: [1.1.9] Watchdog triggered by linear advance calibration pattern

vatral, 2018-11-13:
### Description

Printer: Prusa i3 based IDEX printer on RUMBA board.

Using the calibration pattern from http://marlinfw.org/tools/lin_advance/k-factor.html causes the printer to lock up completely, and trigger the watchdog right after printing the left-most line of the pattern.

I tried disabling the watchdog. The printer indeed locks up, and extruder temperature grows dangerously high. It doesn't appear to be any kind of timing issue.

The printer has printed a full calibration cube on this firmware successfully.

### Steps to Reproduce

1. Print linear calibration pattern

**Expected behavior:** 

It should print

**Actual behavior:** 

Printer locks up hard and triggers the watchdog. All controls become unresponsive, motors and heaters turn off.

#### Additional Information

Video: https://www.youtube.com/watch?v=ygbJfUhfuOs#t=50s
Config and gcode:
[marlin_1_1_9_hang.zip](https://github.com/MarlinFirmware/Marlin/files/2577209/marlin_1_1_9_hang.zip)

vatral, 2018-11-13:
bugfix-2.0.x does print it successfully.

thinkyhead, 2018-11-14:
Marlin 1.1.x is now end-of-life and won't be receiving any more patches. If 2.0.x is working for you then you should stick with it.

closed 2018-11-14 by thinkyhead

thinkyhead, 2018-11-14:
So far we haven't determined that a code bug is involved. The main reason the watchdog can be triggered is trying to do too much computation. This is usually caused by trying to do fast moves on an AVR board in combination with 32x or higher micro-stepping.

The effort required to track down the cause of this issue has to be expended by someone, and right now we are overburdened with our focus on 2.0.x development to help track down such a rare issue with an unknown cause.

I would urge you troubleshoot your setup with vigor until you have found the feature, setting, or combination of settings that exacerbates the issue, and if you are able to narrow it down to some cause that is clearly a bug, perhaps at that point someone may be able to patch it.