MarlinFirmware/Marlin #12424: buttons_reprapworld_keypad not declared with ZONESTAR_LCD

Ludy87, 2018-11-13:
### Description

```
In file included from Marlin\src\Marlin.cpp:33:0:
Marlin\src\lcd/ultralcd.h: In static member function 'static bool MarlinUI::button_pressed()':
Marlin\src\lcd/ultralcd.h:294:21: error: 'buttons_reprapworld_keypad' was not declared in this scope
#define RRK(B) (buttons_reprapworld_keypad & (B))
^
Marlin\src\lcd/ultralcd.h:299:30: note: in expansion of macro 'RRK'
#define BUTTON_CLICK() RRK(EN_REPRAPWORLD_KEYPAD_MIDDLE)
^
Marlin\src\lcd/ultralcd.h:621:50: note: in expansion of macro 'BUTTON_CLICK'
static inline bool button_pressed() { return BUTTON_CLICK(); }
^
*** [.pioenvs\megaatmega2560\src\src\Marlin.cpp.o] Error 1
```
since the introduction of https://github.com/MarlinFirmware/Marlin/pull/12395 the error is issued.

### Steps to Reproduce

1. enable ```#define ZONESTAR_LCD``` [, && select BOARD_RAMPS_14_EFB]
2. compile with PIO
3. read error ;-)

**Expected behavior:** compiled

**Actual behavior:** not compiled

thinkyhead, 2018-11-14:
Should be fixed now following some additional refactoring.

Ludy87, 2018-11-14:
thanks for your work

closed 2018-11-14 by Ludy87