MarlinFirmware/Marlin #12425: Wanhao Duplicator i3 Mini & Rebrands Support

thisiskeithb, 2018-11-13:
### Description

The Wanhao Duplicator i3 Mini & rebrands like the Cocoon Create Model Maker, Monoprice MP i3 Mini (_not to be confused with their Maker Select Mini which is a rebranded Malyan M200_), and PowerSpec Duplicator i3 Mini use a custom all-in-one motherboard/lcd/controller named the "[OnePlus1](https://www.aliexpress.com/item/motherboard-i3mini-0ne-motherboard-New-2017-Wanhao-printer-i3-Mini/32849200836.html)".

I ported over & cleaned up the pins file for the motherboard and a few other settings[V.1.3 source code Wanhao released](https://github.com/garychen99/i3Mini) (Marlin 1.1.4-based) to the [bugfix-1.1.x branch in my Marlin repo](https://github.com/thisiskeithb/Marlin/tree/bugfix-1.1.x-Wanhao-Duplicator-i3-Mini).

If you enable MINIPANEL in base Marlin, the build works, but HAS_LCD_CONTRAST is not set & the default contrast setting is too low, so you can't see the LCD and you can't adjust the contrast with GCODES. I found a few different ways to support this all-in-one board, but creating a `WANHAOI3MINI_DISPLAY` variable like the `CR10_STOCKDISPLAY` seemed to make the most sense. [My proposed changes are here](https://github.com/thisiskeithb/Marlin/commit/83fff06451759ffc2e5bbb718fef2221d9f3a028).

Can these changes be integrated into vanilla Marlin?

#### Additional Information

Check the other branches in my repo for configuration files & bootscreens for the [Cocoon Create Model Maker](https://github.com/thisiskeithb/Marlin/tree/bugfix-1.1.x-Cocoon-Create-Model-Maker/Marlin/example_configurations/Cocoon%20Create/Model%20Maker%20(Wanhao%20Duplicator%20i3%20Mini)) & [Monoprice MP i3 Mini](https://github.com/thisiskeithb/Marlin/tree/bugfix-1.1.x-Monoprice-MP-i3-Mini/Marlin/example_configurations/Monoprice/MP%20i3%20Mini%20(Wanhao%20Duplicator%20i3%20Mini)).

nathanmichaelmoore, 2018-12-06:
Yes! Thank you very much for doing this!

thisiskeithb, 2019-01-02:
Speaking out loud here, using `WANHAOI3MINI_DISPLAY` probably doesn't make sense because the LCD is non-removable.

I still don't know why my ported changes to bugfix-2.0 compile fine, but nothing shows up on the LCD. If anyone has had luck getting this printer working in bugfix-2.0 could chime in, we could create a PR and get this printer (and rebrands) supported.

aeronauta1, 2019-01-11:
Hi , I tried the fix and all works well , but there is no SD card function available , is this ment to be like that?? also I get with the example for the cocoon  I get two errors in the compile report one 
Error while detecting libraries included by C:\Users\jrold\AppData\Local\Temp\arduino_build_825064\sketch\temperature.cpp
and 
Error while detecting libraries included by C:\Users\jrold\AppData\Local\Temp\arduino_build_825064\sketch\ultralcd.cpp
without having changed anything, I am running IDE 1.8.9 hourly build 05:21

any help will be appreciated . Thankyou  

Jorge

aeronauta1, 2019-01-11:
Also tried with the bugfix 2.0 but get too many errors with libraries.. and fails to compile..

Jorge

FREAKYLICIOUSFISHY, 2019-01-14:
eagerly awaiting a solution here this is an issue i now also have this could be an awesome tine board if this gets figured out

aeronauta1, 2019-01-14:
Well, can you specify which problem do you have??..although I have the two red errors on compile the machine works OK...(using 1.1.x)

Jorge

meteozond, 2019-02-12:
@thisiskeithb Just flashed your branch. Works great! But looks like SD detection is not properly implemented. [Pin logic defined but not used.](https://github.com/thisiskeithb/Marlin/blob/bugfix-1.1.x-Wanhao-Duplicator-i3-Mini/Marlin/pins_ONEPLUS1_wanhao.h#L118-L121) As I see there is some additional tricky functionality around SD in original wanhao firmware.

meteozond, 2019-02-12:
Added PR with SD autodetect support for that branch https://github.com/thisiskeithb/Marlin/pull/2/files

aeronauta1, 2019-02-13:
I modified the ultralcd.cpp file as per your changes but on verify I get 

Using library U8glib at version 1.19.1 in folder: C:\Users\jrold\Documents\Arduino\libraries\U8glib 
exit status 1
'IS_SD_PRINTING' was not declared in this scope

aeronauta1, 2019-02-13:
Update , I replaced the line back to IS_SD_PRINTING() and compiled OK , also works fine as it shows now the SD card and detects it OK ... great work Thank you...

Jorge

aeronauta1, 2019-02-13:
Being on a winner I felt more adventuros..I have a Wanhao i3 mini to wich I have added a heated bed ( Mosfet and silicon mat) using the connector labeled BEDDADD , all works well and the bed heats up  and keeps the temp OK without PID  (//#define PIDTEMPBED  ).
My problem is that when I enable #define PIDTEMPBED and verify all is well , as I upload the up;oad works ok but it does not display anything in the LCD , also I can connect to the printer with Octoprint but the controls do not move the axis and sometime he steppers start making a bad noise and nothing happens.  (it is not related to the ultralcd.cpp update as I changed back to the original) , can anyone pleas give a suggestion on a solution??

Thankyou...Jorge

thisiskeithb, 2019-02-14:
Unless someone can get this printer fully working in bugfix2.0, there’s not much of a reason to keep this issue open.

closed 2019-02-14 by thisiskeithb

thisiskeithb, 2019-07-09:
For those of you still watching this issue, here's my PR with support for this printer & rebrands: https://github.com/MarlinFirmware/Marlin/pull/14559.

reallyallnamestaken, 2019-08-11:
I installed the latest version and still no LCD display.
Comparing against Wanhao provided code (1.1.x) there is "ultralcd3_WanhaoOnePlus.cpp", which seem to handle the contrast correctly. This is not present in the main Marlin code. Any chance including it would fix these issues?

https://github.com/garychen99/i3Mini

thisiskeithb, 2019-08-11:
It *was* working, but I just flashed mine with the latest commit & can confirm the contrast is not set properly.

For the time being, you can send `M250 C255` to set the contrast while I look into it.