MarlinFirmware/Marlin #12427: v_allowable为什么这么计算？

shiqingfeng, 2018-11-14:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-11-14:
Marlin 2.0 uses `v_allowable_sqr` which requires less calculation.

closed 2018-11-14 by thinkyhead

shiqingfeng, 2018-11-21:
需要较少的计算是什么意思？可以给出更详细的解释吗？