MarlinFirmware/Marlin #12436: [bugfix 2.0.x] Re-Arm with LCD and SD Card reader

ant0nyk1ng, 2018-11-15:
Re-Arm using the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER with the SD Card enabled

`#define SDSUPPORT`

The MiniSD Card on the Re_Arm board is accessed and not the SD Card on the LCD Smart Controller.

If `//#define SDSUPPORT` then no SD Card is accesable by the LCD.

I am not sure if this is a bug or a feature and I am have another step to enable the SD card I want to use?

If it's a case this is a bug then maybe some people might want to be able to have this access... unless there is a reason you shouldn't use the MiniUSB on the Re-Arm motherboard of course.

kAdonis, 2018-11-15:
It's a feature 
To use the SD Card from REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER you need to edit your pins_RAMPS_RE_ARM.h:

```
#define LPC_SD_LCD          // Marlin uses the SD drive attached to the LCD
//#define LPC_SD_ONBOARD        // Marlin uses the SD drive on the control board
```

gloomyandy, 2018-11-15:
One thing you may want to think about is if you plan to use USB access to the SD card. If you keep the default configuration then the SD card on the main board is available to both your PC (via USB) and to Marlin. This makes it really easy to slice on your PC and save the gcode to the printers SD card, then you can just print it from the SD card. If you enable the SD card on the LCD then the SD card on the main board is still available to the PC (and can be used to update the firmware), but you can no longer print from it (as Marlin only allows for a single SD card, and that is now the one on the LCD). This may or may not matter to you.

ant0nyk1ng, 2018-11-16:
> 
> 
> It's a feature
> To use the SD Card from REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER you need to edit your pins_RAMPS_RE_ARM.h:
> 
> ```
> #define LPC_SD_LCD          // Marlin uses the SD drive attached to the LCD
> //#define LPC_SD_ONBOARD        // Marlin uses the SD drive on the control board
> ```

Thanks, I didn't see that and wasn't aware. I have it working.

gloomyandy, thanks as well.

closed 2018-11-16 by ant0nyk1ng