MarlinFirmware/Marlin #12437: [2.0.x] M24 requires M108 to resume, M25 no longer works. Why?

marcio-ao, 2018-11-15:
Our LCD controller when doing a USB print, if the user hits the pause button, it enqueues "M24". When the user hits the resume button it enqueues "M25". This used to work perfectly. However, now when I try to resume a print, it does not work. Instead it reports:

```
echo:Send M108 to resume
```

Why is this? This seems like a bug.

marcio-ao, 2018-11-15:
Googling through the code, it looks like this maybe related to a filament change. But M125 is supposed to park the head, not initiate a filament change.

For the record, I would also like to use "M24" as my FILAMENT_RUNOUT_SCRIPT. The thing is that the filament change routines in Marlin are tied pretty closely to the UltraLCD and would be incompatible with our touch controller, so all I want to do on a filament runout is pause the print and park the head, and then resume the print upon "M25". The filament change will be done manually by the user by clicking buttons on the UI.

marcio-ao, 2018-11-15:
Okay, it looks like the ADVANCED_PAUSE_FEATURE was the problem. By disabling that I get the normal behavior of "M24" or "M25". It looks like I have to enqueue "M24\nM27" to get the park behavior, but that is easily done.

For what it is worth, I think the ADVANCED_PAUSE feature is overly complex. Rather than being a implemented as a complex GCODE that calls all sorts of hooks into the UltraLCD code, I think a better approach would be to have the filament change be handled by the UI code in whatever manner was appropriate for that interface, such as by enqueueing normal motion GCODE commands in response to user interaction. M600 should not implement a filament change, but instead defer to the UI code to do whatever is appropriate for that implementation.

marcio-ao, 2018-11-15:
I guess when I enqueue a G27, the printer does not return to the proper position on resume and instead air prints. This is what the code looks like in extensible_ui:

```
  void pausePrint() {
    #if ENABLED(SDSUPPORT)
      card.pauseSDPrint();
      print_job_timer.pause();
      #if ENABLED(NOZZLE_PARK_FEATURE)
        //if (!axis_unhomed_error()) {
        //  planner.synchronize();
        //  Nozzle::park(0);
        //}
        enqueueCommands(F("G27"));
      #endif
      ExtUI::onStatusChanged(PSTR(MSG_PRINT_PAUSED));
    #endif
  }
```

Any thoughts?

thinkyhead, 2018-11-16:
Hmm, when there's an LCD controller, the wait message should say "`Press button (or M108) to resume`." If Extensible UI provides a method whereby it can receive a button click, then that method should be applied in `M125` to compliment the current method which looks for `HAS_LCD_MENU` to determine that there's a controller which can provide a click.

thinkyhead, 2018-11-16:
See the `wait_for_confirmation` function in `pause.cpp`.

tcm0116, 2018-11-16:
>I think a better approach would be to have the filament change be handled by the UI code in whatever manner was appropriate for that interface

How would this work for printers without an LCD? M600 should still work.

marcio-ao, 2018-11-16:
> Hmm, when there's an LCD controller, the wait message should say "Press button (or M108) to resume."

The current filament change routine in UltraLCD is quite complex. It currently invokes a sequence of linear steps like so:

- The user is presented with a menu of extruders.
- The user is presented with a set of pre-heat options.
- The user sees an screen telling them to wait while the extruder pre-heats/unload.
- The user is asked to press a button.
- The user sees a screen telling them to wait for the filament to load.
- The user sees a screen asking them whether to purge more filament or continue.
- The user sees a screen asking them to wait for the filament change to complete.

Overall, this is a very linear sequence and I believe this is too restrictive. In the case of the filament screen I implemented:

- The user has the option to switch between tool heads on-the-fly and initiate pre-heat on them simultaneously.
- They have the option to advance or retract filament, either "momentarily" when holding down a button or "continuously" without having to hold down the button.

Here is what the screen currently looks like:

[image]

My concern is that if Marlin dictates that all UIs will implement a strictly linear sequence for a filament change using a single button, it eliminates much of the advantages of having an advanced multi-button touch screen.

thinkyhead, 2018-11-18:
`M600` was originally designed to be used as a complete filament change, generally for headless systems, and was the first and only filament change feature implemented so far.

To do a more granular user interface, you can borrow some of the settings from the `ADVANCED_PAUSE_FEATURE` and use them in the ExtUI (or not) and implement the various functions in your "Extruder Selection" screen as simpler operations, such as `G1 E` moves, and so on. For that user interface, `M600` would not be appropriate. The appropriate thing to do for an external controller would be to implement a custom G-code that invokes the custom UI and set that as the `FILAMENT_RUNOUT_SCRIPT` instead of `M600`.

closed 2019-02-14 by marcio-ao