MarlinFirmware/Marlin #12438: [2.0.x] Disable hotend via M-code

ghulands, 2018-11-15:
### Description

I have a Formbot T-Rex 2 that has dual hotends that can be enabled or disabled by quickly unclamping it from the X-axis belt. A lot of the time I am only using the first extruder and leave the second one parked off to the side unclamped. I just started testing the 2.0.x firmware and got the Heating Failed message for the extruder that was unplugged (T1).

It would be good if a hotend can be disabled via an M-code so that it is not part of the temperature protection if it's disabled for that job.

ghulands, 2018-11-15:
Tagging @Roxy-3D as they have a T-Rex 2+

silentninja1, 2018-11-15:
The 2+ is totally different from the 2. They use different boards, 2+ is idex so each is on independent belts, ect. Roxy does not have a older 2 with the MKS Gen 1.4 board.

A more complete consideration for this case is to disable monitoring of a hotend that is not on, as in if temperature is 0 exclude from watchdog. This would be relevant to any multiple hotend system. Im not sure if theres any other issues in implementing such a thing though so ill let somebody who knows that section of code comment on if its feasible or not.

I can see it is something ive seen while working on a few machines, just didnt bother me enough to do anything about.

ghulands, 2018-11-15:
The temperature reported is -15 for this printer when disconnected. Not sure if that is the same for all machines.

silentninja1, 2018-11-15:
Thats based on the temp table for open circuit for enabled sensor. Not relevant to how I would look to implement though, which is ignore unless a non-0 target temperature is set.

ghulands, 2018-11-15:
Sounds like a better approach than disabling via M-code. Updated title.

TheSFReader, 2018-11-16:
Detail : I think an overhigh temp error should still be raised in the case of no target

GMagician, 2018-12-05:
@ghulands you may close this, feature has been added

boelle, 2019-03-13:
@thinkyhead what @GMagician said ^^^^

InsanityAutomation, 2019-05-25:
Agreed @thinkyhead this one is done and can be closed.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this is done

closed 2019-10-15 by boelle