MarlinFirmware/Marlin #12441: [bugfix 2.0.x] [Q] G29 Is it correct to bring the bed and extruder temperature to 0?

mr-miky, 2018-11-15:
### Description

At the end of G29, the temperature of bed and etruder are set to 0

### Steps to Reproduce

1. M190 S60 ; set bed temperature and wait for it to be reached
2. M104 S195 ; set temperature
3. G29		    ;bed level

**Expected behavior:** 
Bed levelling, bed and extruder ready to print at the right temperature

**Actual behavior:** [What actually happens]
Bed levelled , extruder & bed temperature set to 0
#### Additional Information

mr-miky, 2018-11-17:
After several tests looking at the communication log I realized that octoprint sends M104 S0 and M140 S0.
The thing that drove me crazy is that even if octoprint is not printing it sends those commands therefore even if the leveling is started from the LCD display menu at the end of the leveling octoprint sends the command to shut down the extruder and bed.

It remains only to understand why octoprint behaves like that but it is not a marlin problem.

mr-miky, 2018-11-17:
And the winner is the "Auto bed leveling expert" plugin that sets the temperature of the extruder and the bed to 0 at the end of G29 even if it has not started the leveling procedure.

bvalat, 2018-11-27:
Many thanks! Had the very same problem last night.
Since i installed my bl sensor last night , i was wondering if i made a mistake in my marlin conf/compilation.
I reverted back to non ABL firmware and everithing was OK.
Now you mention that plugin, i installed it.
I will try without this plugin tonight.

thinkyhead, 2018-11-27:
Please let the OctoPrint project know. Closing since there's nothing to do.

closed 2018-11-27 by thinkyhead