MarlinFirmware/Marlin #12442: EEPROM Read problem

setar, 2018-11-16:
bugfix-2.0.x , 7a075a5dc99b8769061b197854af70aefccbdae8

my config : https://github.com/setar/Marlin/tree/bugfix-2.0.x-3ntProf_v10_AzteegX5_short_Yhead

EEPROM Read problem:
```
SENDING:M502
echo:Hardcoded Default Settings Loaded
SENDING:M500
echo:Settings Stored (589 bytes; crc 19039)
SENDING:M501
echo:EEPROM version mismatch (EEPROM=? Marlin=V62)
echo:Hardcoded Default Settings Loaded
```
for debug i try define DEBUG_EEPROM_READWRITE but no result.
Any idea for debugging ?

gloomyandy, 2018-11-16:
Have you been using Marlin before on this board? If so how old was the version you previously had? Did the eeprom work then? Have you had other software installed on the board?

Can you try the following.... Send M502 followed by M500 repeat this at least 8 times and then try an M501.

setar, 2018-11-16:
The board (Azteeg_X5_GT) work fine on dc1113165601db4675036f52267f8489464107a1 (Tue Sep 25 2018)

gloomyandy, 2018-11-16:
Can you go back to that older version and edit the persistent_store_api.h in the LPC1768 HAL to enable the use of flash for eeprom storage then try things again? You should get some debug output.

setar, 2018-11-16:
ok:
(i try with more older commit 3d7f2c1fde5d93f55b77a0521363f94a4047133e Thu Sep 06 2018
new version need additional time for check)
work both - SD and EEPROM store:
```
SENDING:M502
echo:Hardcoded Default Settings Loaded
SENDING:M500
Blank check status: 8
Flash slot: 6
Address: 516096
echo:Settings Stored (691 bytes; crc 8294)
Writing data to: 5
CopyRAM2Flash status: 0
SENDING:M501
Blank check status: 8
Flash slot: 5
Address: 512000
Blank check status: 8
Flash slot: 5
Address: 512000
echo:V56 stored settings retrieved (691 bytes; crc 8294)
Unified Bed Leveling System v1.01 inactive.
Unified Bed Leveling initialized.
UBL System reset()
  G21    ; Units in mm (mm)
  M149 C ; Units in Celsius
  M200 D1.75
  M200 T1 D1.75
  M200 D0
  M92 X106.67 Y106.67 Z960.00 E225.00
  M203 X150.00 Y150.00 Z18.00 E10.00
  M201 X1000 Y1000 Z60 E500
  M204 P1000.00 R500.00 T1000.00
  M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
  M206 X0.00 Y0.00 Z0.00
  M218 T1 X0.00 Y0.00 Z0.000
  M420 S0 Z10.00
  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
  M301 E0 P16.42 I1.07 D63.01
  M301 E1 P16.42 I1.07 D63.01
  M304 P165.01 I16.50 D412.47
  M250 C140
  M851 Z-0.30
```

AND current 7a075a5dc99b8769061b197854af70aefccbdae8  
try read old EEPROM data:
```
SENDING:M502
echo:Hardcoded Default Settings Loaded
SENDING:M500
echo:Settings Stored (565 bytes; crc 19379)
SENDING:M501
echo:EEPROM version mismatch (EEPROM=V56 Marlin=V62)
echo:Hardcoded Default Settings Loaded
```
i think problem with EEPROM write

AND current work fine with SD store ...
```
SENDING:M502
echo:Hardcoded Default Settings Loaded
SENDING:M500
echo:Settings Stored (565 bytes; crc 19379)
SENDING:M501
echo:V62 stored settings retrieved (565 bytes; crc 19379)
Unified Bed Leveling System v1.01 inactive.
Unified Bed Leveling initialized.
UBL System reset()
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X106.67 Y106.67 Z1920.00 E280.00
echo:Maximum feedrates (units/s):
echo:  M203 X150.00 Y150.00 Z15.00 E100.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z60.00 E1000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.00 R500.00 T1000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S0 Z10.00
Unified Bed Leveling System v1.01 inactive.
Active Mesh Slot: -1
EEPROM can hold 16 meshes.
echo:Servo Angles:
echo:  M281 P0 L10 U90
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P16.42 I1.07 D63.01
echo:  M304 P165.01 I16.50 D412.47
echo:LCD Contrast:
echo:  M250 C140
echo:Z-Probe Offset (mm): (mm):
echo:  M851 Z-0.70
echo:Stepper driver current:
echo:  M906 X1200 Y1200 Z1400
  M906 T0 E1200

```

gloomyandy, 2018-11-16:
Unfortunately I can't debug this at the moment. Is there any chance you can copy the file persistent_store_flash.cpp from the older build (the one that works and that produces debug) and use that version of the file in the current version of Marlin? That way we should see the write debug from the HAL (if there is any).

gloomyandy, 2018-11-17:
OK I've just spent a couple of hours trying to reproduce this and nothing I can do will make my LPC1768 based SBase board do anything other than work fine. Having said that this is the 2nd report I've seen in which LPC1769 based systems have some sort of issue with storing eeprom in flash. The other report only had issues with storing a mesh not with storing the other settings. I have a feeling that if there is a problem it is somehow timing related. The only slightly odd thing I've seen is that the framework actually calls the PrepareSector function for you so the calls in the Marlin code are not really needed, perhaps doing this operation twice is causing some issue? So can you try commenting out the calls to PrepareSector in the Marlin code and rebuilding.

@p3p do you have an LPC1769 based board? Any chance you can try and reproduce this problem?

@thinkyhead While investigating this I noticed a couple of odd uses of the persistent store API by the configuration store code. The first issue is that in load_mesh there is the following code:
``` C
        int pos = mesh_slot_offset(slot);
        uint16_t crc = 0;
        uint8_t * const dest = into ? (uint8_t*)into : (uint8_t*)&ubl.z_values;

        persistentStore.access_start();
        const uint16_t status = persistentStore.read_data(pos, dest, sizeof(ubl.z_values), &crc);
        persistentStore.access_finish();

        if (status)
          SERIAL_PROTOCOLPGM("?Unable to load mesh data.\n");

        #if ENABLED(EEPROM_CHITCHAT)
          else
            SERIAL_PROTOCOLLNPAIR("Mesh loaded from slot ", slot);
        #endif
        EEPROM_FINISH();
```
It seems to me that this is incorrect as the EEPROM_FINISH will be calling persistentStore.access_finish(); which has already been called. So you end up with a finish being called without the matching start.

The second issue is that when UBL is active and there is an active mesh then a call is made to the above load_mesh code to load the active mesh. However this call is made inside of the EEPROM_START, EEPROM_FINISH pair, this creates a nested call to the persistent store api, is it intended that this is valid? It is noticeable that the code that writes the data to eeprom takes care to avoid this nested call. So the sequence is on write...
``` C
persistentStore.access_start();
// write main config
persistentStore.access_finish(); 
persistentStore.access_start();
// save mesh
persistentStore.access_finish();
```

But on read it looks like...
``` C
persistentStore.access_start();
// read main config
persistentStore.access_start();
// read mesh
persistentStore.access_finish();
persistentStore.access_finish();
```

Should this second form be valid? At the moment it does not cause a problem for the LPC1768 read code, however if the same pattern was used for the write code it probably would.

stefan85, 2018-11-19:
I have exactly the same issue.

I Tried the same thing the commit of Sep 06 2018 and it works fine, but the latest one does not. I use the SD card as eeprom. Tomorrow I will try commits in between to narrow it down.

> ok:
> (i try with more older commit [3d7f2c1](https://github.com/MarlinFirmware/Marlin/commit/3d7f2c1fde5d93f55b77a0521363f94a4047133e) Thu Sep 06 2018
> new version need additional time for check)
> work both - SD and EEPROM store:
> 
> ```
> SENDING:M502
> echo:Hardcoded Default Settings Loaded
> SENDING:M500
> Blank check status: 8
> Flash slot: 6
> Address: 516096
> echo:Settings Stored (691 bytes; crc 8294)
> Writing data to: 5
> CopyRAM2Flash status: 0
> SENDING:M501
> Blank check status: 8
> Flash slot: 5
> Address: 512000
> Blank check status: 8
> Flash slot: 5
> Address: 512000
> echo:V56 stored settings retrieved (691 bytes; crc 8294)
> Unified Bed Leveling System v1.01 inactive.
> Unified Bed Leveling initialized.
> UBL System reset()
>   G21    ; Units in mm (mm)
>   M149 C ; Units in Celsius
>   M200 D1.75
>   M200 T1 D1.75
>   M200 D0
>   M92 X106.67 Y106.67 Z960.00 E225.00
>   M203 X150.00 Y150.00 Z18.00 E10.00
>   M201 X1000 Y1000 Z60 E500
>   M204 P1000.00 R500.00 T1000.00
>   M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
>   M206 X0.00 Y0.00 Z0.00
>   M218 T1 X0.00 Y0.00 Z0.000
>   M420 S0 Z10.00
>   M145 S0 H180 B70 F0
>   M145 S1 H240 B110 F0
>   M301 E0 P16.42 I1.07 D63.01
>   M301 E1 P16.42 I1.07 D63.01
>   M304 P165.01 I16.50 D412.47
>   M250 C140
>   M851 Z-0.30
> ```
> AND current [7a075a5](https://github.com/MarlinFirmware/Marlin/commit/7a075a5dc99b8769061b197854af70aefccbdae8)
> try read old EEPROM data:
> 
> ```
> SENDING:M502
> echo:Hardcoded Default Settings Loaded
> SENDING:M500
> echo:Settings Stored (565 bytes; crc 19379)
> SENDING:M501
> echo:EEPROM version mismatch (EEPROM=V56 Marlin=V62)
> echo:Hardcoded Default Settings Loaded
> ```
> i think problem with EEPROM write
> 
> AND current work fine with SD store ...
> 
> ```
> SENDING:M502
> echo:Hardcoded Default Settings Loaded
> SENDING:M500
> echo:Settings Stored (565 bytes; crc 19379)
> SENDING:M501
> echo:V62 stored settings retrieved (565 bytes; crc 19379)
> Unified Bed Leveling System v1.01 inactive.
> Unified Bed Leveling initialized.
> UBL System reset()
> echo:  G21    ; Units in mm (mm)
> echo:  M149 C ; Units in Celsius
> echo:Filament settings: Disabled
> echo:  M200 D1.75
> echo:  M200 D0
> echo:Steps per unit:
> echo:  M92 X106.67 Y106.67 Z1920.00 E280.00
> echo:Maximum feedrates (units/s):
> echo:  M203 X150.00 Y150.00 Z15.00 E100.00
> echo:Maximum Acceleration (units/s2):
> echo:  M201 X1000.00 Y1000.00 Z60.00 E1000.00
> echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
> echo:  M204 P1000.00 R500.00 T1000.00
> echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
> echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
> echo:Home offset:
> echo:  M206 X0.00 Y0.00 Z0.00
> echo:Unified Bed Leveling:
> echo:  M420 S0 Z10.00
> Unified Bed Leveling System v1.01 inactive.
> Active Mesh Slot: -1
> EEPROM can hold 16 meshes.
> echo:Servo Angles:
> echo:  M281 P0 L10 U90
> echo:Material heatup parameters:
> echo:  M145 S0 H180 B70 F0
> echo:  M145 S1 H240 B110 F0
> echo:PID settings:
> echo:  M301 P16.42 I1.07 D63.01
> echo:  M304 P165.01 I16.50 D412.47
> echo:LCD Contrast:
> echo:  M250 C140
> echo:Z-Probe Offset (mm): (mm):
> echo:  M851 Z-0.70
> echo:Stepper driver current:
> echo:  M906 X1200 Y1200 Z1400
>   M906 T0 E1200
> ```

gloomyandy, 2018-11-19:
If you use the SD card as eeprom you will not see the problem. The 6th September version uses the SD card by default, the later build will use flash memory. So if you just want to get going you can change the setting in the persistent_store_api.h file of the current build.  If you want to help and debug this issue then please try the following...
1. Using the 6th september version edit the persistent_store_api.h file and uncomment #define FLASH_EEPROM and try things again and post the output.
2. Try copying the persistent_store_flash.cpp file from the 6th Sept version over to the most recent build and try things and post the output. 
3. Try commenting out all of the calls to the PrepareSector function in the most recent version of persistent_store_flash.cpp and try it again.

Thanks for your help.

stefan85, 2018-11-19:
So default of current build is flash memory. How can I let it use the sd card in the current build? Maybe I can check this first. Then I can try to reproduce your described steps.

gloomyandy, 2018-11-19:
See the comment above. You need to comment out the line #define FLASH_EEPROM in the file persistent_store_api.h in the HAL_LPC1768 directory.

setar, 2018-11-20:
> If you use the SD card as eeprom you will not see the problem. The 6th September version uses the SD card by default, the later build will use flash memory. So if you just want to get going you can [...]

3. not work:
```
SENDING:M501
Blank check status: 8
Flash slot: 4
Address: 507904
echo:EEPROM version mismatch (EEPROM=V56 Marlin=V63)
echo:Hardcoded Default Settings Loaded
SENDING:M502
echo:Hardcoded Default Settings Loaded
SENDING:M500
Blank check status: 8
Flash slot: 4
Address: 507904
echo:Settings Stored (566 bytes; crc 30332)
Writing data to: 3
CopyRAM2Flash status: 2
SENDING:M500
Blank check status: 8
Flash slot: 4
Address: 507904
echo:Settings Stored (566 bytes; crc 30332)
Writing data to: 3
CopyRAM2Flash status: 2
SENDING:M500
Blank check status: 8
Flash slot: 4
Address: 507904
echo:Settings Stored (566 bytes; crc 30332)
Writing data to: 3
CopyRAM2Flash status: 2
SENDING:M500
Blank check status: 8
Flash slot: 4
Address: 507904
echo:Settings Stored (566 bytes; crc 30332)
Writing data to: 3
CopyRAM2Flash status: 2
SENDING:M502
echo:Hardcoded Default Settings Loaded
SENDING:M501
Blank check status: 8
Flash slot: 4
Address: 507904
echo:EEPROM version mismatch (EEPROM=V56 Marlin=V63)
echo:Hardcoded Default Settings Loaded
```

gloomyandy, 2018-11-20:
The debug output is interesting, looks like there may be an alignment issue can you edit the persistent_store_flash.cpp file and change the following line...
``` C
static uint8_t ram_eeprom[EEPROM_SIZE];
```
to
``` C
static uint8_t ram_eeprom[EEPROM_SIZE] __attribute__((aligned(4))) = {0};
```

stefan85, 2018-11-20:
Setar, do you have a LPC1769 or LPC1768 board. I have a LPC1769 board and with all builds I get this message: echo:EEPROM version mismatch (EEPROM=?

I tried all suggested options and get the same result. So maybe it is an issue with the LPC1769

gloomyandy, 2018-11-20:
@stefan85 Did you try the change I've just posted above (the code change that adds data alignment)? Did you try using the older version of the persistent_store_flash.cpp file (the one with extra diagnostics in it)? If you did please post your output. 

Note that you will get the error you mention until you actually store a valid config in flash. So be sure to test both a write operation and a read operation.

stefan85, 2018-11-20:
That was really strange. I did the following steps:

- I copied the old persistent_store_flash.cpp. 
- Did an M504 with this result:  echo:EEPROM version mismatch (EEPROM=? Marlin=V63)   This makes sense because the flash should be empty
- Followed with M500
- M504 gave me echo:EEPROM OK (this never worked on the new persistent_store_flash.cpp)
-Then I switched to the new persistent_store_flash.cpp and everything works fine now. Very strange.

I don't know if it is any help but it seems to be issues with first use of the eeprom in my case.

gloomyandy, 2018-11-20:
Actually I suspect that your new version is not working (or if it is it is just down to luck because the alignment of the ram buffer just happens to end up correct). Have you tried actually changing some of the eeprom contents and checking that the new values are correct after storing and loading? The problem in the logs above is during a write operation and results in the existing content not being changed. This error is not reported by the current Marlin build and so you may not realise that the new values have not been saved.

setar, 2018-11-21:
> Setar, do you have a LPC1769 or LPC1768 board. I have a LPC1769 board and with all builds I get this message: echo:EEPROM version mismatch (EEPROM=?
> 
> I tried all suggested options and get the same result. So maybe it is an issue with the LPC1769

Azteeg X5 GT by design have LPC1769, but for build i use pio run -e LPC1768 -t upload

setar, 2018-11-21:
> Can you try changing a value in the framework for me:
> `USER/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/system/system_LPC17xx.c`
> line 325: `#define FLASHCFG_Val 0x0000303A`
> change to
> `#define FLASHCFG_Val 0x0000403A`
> 
> I'm not sure this will help as the system files are almost direct copies from smoothieware but the current value seems to be wrong only supporting upto 80Mhz, The manual specifies 4 (5 ticks) should work at 120Mhz as well as the 100Mhz noted in the comment.

not work:
```
SENDING:M501
echo:EEPROM version mismatch (EEPROM=V56 Marlin=V63)
echo:Hardcoded Default Settings Loaded
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M501
echo:EEPROM version mismatch (EEPROM=V56 Marlin=V63)
echo:Hardcoded Default Settings Loaded
Setting bed temperature to 120.000000 degrees Celsius.
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M501
echo:EEPROM version mismatch (EEPROM=V56 Marlin=V63)
echo:Hardcoded Default Settings Loaded
```

setar, 2018-11-21:
> The debug output is interesting, looks like there may be an alignment issue can you edit the persistent_store_flash.cpp file and change the following line...
> 
> ```c
> static uint8_t ram_eeprom[EEPROM_SIZE];
> ```
> 
> to
> 
> ```c
> static uint8_t ram_eeprom[EEPROM_SIZE] __attribute__((aligned(4))) = {0};
> ```

Bingo!

```
SENDING:M501
echo:EEPROM version mismatch (EEPROM=V56 Marlin=V63)
echo:Hardcoded Default Settings Loaded
SENDING:M502
echo:Hardcoded Default Settings Loaded
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M500
echo:Settings Stored (566 bytes; crc 30332)
SENDING:M501
echo:V63 stored settings retrieved (566 bytes; crc 30332)
Unified Bed Leveling System v1.01 inactive.
Unified Bed Leveling initialized.
UBL System reset()
--- skip ---
```

check
```
SENDING:M851 Z-1.75
SENDING:M500
echo:Settings Stored (566 bytes; crc 16102)
SENDING:M501
echo:V63 stored settings retrieved (566 bytes; crc 16102)
Unified Bed Leveling System v1.01 inactive.
Unified Bed Leveling initialized.
UBL System reset()
--- skip ---
echo:  M851 Z-1.75
--- skip ---
```

setar, 2018-11-21:
UBL Mesh store to EEPROM also work fine:
```
SENDING:G29 P3 T
Bed Topography Report:
    (  0,390)                                      (390,390)
        0       1       2       3       4       5       6
 6 | +2.752  +2.763  +2.804  +3.183  +3.091  +3.012  +3.036
   |
 5 | +3.408  +3.319  +3.216  +3.139  +3.219  +3.297  +3.302
   |
 4 | +4.204  +4.041  +4.090  +3.980  +3.767  +3.884  +3.977
   |
 3 | +4.650  +4.549  +4.512  +4.536  +4.490  +4.198  +4.614
   |
 2 | +5.074  +4.994  +4.938  +4.948  +4.984  +4.858  +4.991
   |
 1 | +5.739  +5.602  +5.455  +5.647  +5.606  +5.549  +5.697
   |
 0 |[+6.405] +6.210  +5.972  +6.347  +6.228  +6.241  +6.402
        0       1       2       3       4       5       6
    (  0,  0)                                      (390,  0)
>>> g29 s1
SENDING:G29 S1
Mesh saved in slot 1
Done.
>>> g29 l1
SENDING:G29 L1
Mesh loaded from slot 1
Done.
>>> g29 t
SENDING:G29 T
Bed Topography Report:
    (  0,390)                                      (390,390)
        0       1       2       3       4       5       6
 6 | +2.752  +2.763  +2.804  +3.183  +3.091  +3.012  +3.036
   |
 5 | +3.408  +3.319  +3.216  +3.139  +3.219  +3.297  +3.302
   |
 4 | +4.204  +4.041  +4.090  +3.980  +3.767  +3.884  +3.977
   |
 3 | +4.650  +4.549  +4.512  +4.536  +4.490  +4.198  +4.614
   |
 2 | +5.074  +4.994  +4.938  +4.948  +4.984  +4.858  +4.991
   |
 1 | +5.739  +5.602  +5.455  +5.647  +5.606  +5.549  +5.697
   |
 0 |[+6.405] +6.210  +5.972  +6.347  +6.228  +6.241  +6.402
        0       1       2       3       4       5       6
    (  0,  0)                                      (390,  0)

```

alexxy, 2018-11-21:
Cool. Seems like this also should solve #12407

setar, 2018-11-21:
ок, `pio run -e LPC1769 -t upload` too work fine, next time i will use this command

gloomyandy, 2018-11-21:
Thanks for testing this I'll put together a PR for the changes later today.

setar, 2018-11-23:
fix work fine, thanks

closed 2018-11-23 by setar