MarlinFirmware/Marlin #12443: PID for Bed Enhancement

Cr0nJ0b, 2018-11-16:
### Description

Feature Enhancement Request:  PIDTEMPBED

For my Anet Modified printer I notice that when selecting PIDTEMPBED the temp takes an inordinately long time to settle.  I believe that the code is likely pulsing up to to correct temp and as it gets closer and closer to the final temp, it regulates more and more and just takes really long to get the "perfect temp"  I've tested with PIDTEMPBED and BED_LIMIT_SWITCHING.  PID takes like 10min to get to temp, while BLS takes only about 5 min.  I think if the code were set to say force temp on (not pulse) until you reach like temp +1C, and then turn on pulse that this would speed up the warm up and still work fine.

### Steps to Reproduce

1. Set firmware to BED_LIMIT_SWITCH
2. Warm up to 200C from cold start and time
3. Cool down and switch to PIDTEMPBED
4,  Warm up to 200C from cold start and time this.

**Expected behavior:** I would expect that these two settings would take about the same amount of time to reach bed 200C

**Actual behavior:** PIDTEMPBED takes nearly 2 x longer for me.

#### Additional Information

AnHardt, 2018-11-16:
Configs please.

RudyBenner, 2018-11-17:
"borrow" a more aggressive set of PID numbers from someone else, use them to get you started and develop your own.
echo:PID settings:

M304 P225.39 I39.30 D323.12

Enter into your terminal.

Cr0nJ0b, 2018-11-17:
I'll try to PID numbers, but I did a PID_Tune once I setup PIDTEMPBED.  It just seems like the pulsing takes a great deal longer than the bed_limit to reach temp.

Configs attached.

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/2592085/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/2592086/Configuration.h.txt)

RudyBenner, 2018-11-17:
I will have look when I get home. Does not sound right.

RudyBenner, 2018-11-17:
you have it bass ackwards.

[image]

You want 

#define PIDTEMPBED

//#define BED_LIMIT_SWITCHING

RudyBenner, 2018-11-17:
This is what it should look like. Temperature on the top, power on the bottom.

[image]

RudyBenner, 2018-11-17:
Insulation under the bed helps. I can make a suggestion, but not here.

Cr0nJ0b, 2018-11-18:
Thanks for the response. I understand the the configs I posted had the PIDBEDTEMP off.  That's because of the issue.  It's interesting that your graph shows full power up to temp.  I was looking at the LED light on the bed and it flashes with PIDBEDTEMP set and stays solid with BED_LIMIT_SWITCH.  What software are you using for this?  Also, the cork suggestion is good.  I'll look into that for efficiency sake.

thanks

RudyBenner, 2018-11-18:
No, not cork. Filter material that goes in a return air duct. Using Marlin. There are also some settings in configuration_adv.h related. Best thing to do is not mess with them.

RudyBenner, 2018-11-18:
BED_LIMIT_SWITCH just checks the temperature at regular intervals.

thinkyhead, 2018-11-22:
Indeed. With `PIDTEMPBED` the power should be at full (100% DC) until it gets close to the target, at which point PID will take over. Just as an experiment, see if it makes any difference to set `MAX_BED_POWER` to 254 or 253.

RudyBenner, 2018-11-22:
#define BANG_MAX 220 //hotend
#define MAX_BED_POWER 255 //hotbed

I run the hotend at a lower setting. Makes things more manageable.

boelle, 2019-02-21:
@Cr0nJ0b problem solved?

Cr0nJ0b, 2019-02-21:
I never solved the issue per se.  I just shut off PIDBEDTEMP and use the BED_LIMIT_SWITCH.  It works for me.

boelle, 2019-02-22:
@Cr0nJ0b click green close button below

closed 2019-02-22 by Cr0nJ0b