MarlinFirmware/Marlin #12447: TMC2660 enable doesn't get pulled low

stefan85, 2018-11-16:
I made a board with an LPC1769 in combination with TMC2660 drivers (REARM pinning). After debugging I found that the enable lines of the drivers do not get pulled to GND to enable.

When I connect the enable pin to GND, the driver works fine. I can turn the motors off with M84. 
But somehow, the microcontroller doesn't control these pins. Is this a know issue, or is there someone else using this hardware combination?

kAdonis, 2018-11-17:
@teemuatlut wrote in https://github.com/MarlinFirmware/Marlin/pull/11998#issue-220185771 :

> The drivers do not have an EN pin so the function is implemented in software through SPI, very similar to how TMC26x is not at the moment. The TMC2660 is based on 26x after all.

boelle, 2019-02-21:
@stefan85 problem solved?

stefan85, 2019-02-21:
Have not checked with latest firmware. Will try it this weekend

boelle, 2019-02-21:
remember to use bugfix 2.0

stefan85, 2019-02-24:
Problem is sovled, I can use software and hardware enable.

closed 2019-02-24 by stefan85

boelle, 2019-02-24:
just curious, what was the fault?

stefan85, 2019-02-24:
In a older version I used, I could only use software enable. But in later commit this was fixed. I tried it yesterday