MarlinFirmware/Marlin #12454: [2.0.x]   Reprap discount full graphic smart controller

hamster65, 2018-11-17:
I undefined LCD_TIMEOUT_TO_STATUS  in configuration_adv because I don't want the lcd to return to the status screen after a while. This worked with marlin 1.8 but with the current 2.0 branch, the lcd returns to the status screen after some seconds. Do I have to set LCD_TIMEOUT_TO_STATUS to 0 to disable this feature?

I also noticed that during homing, there are to decimals displayed at the z axis. The second decimal does not fit on the lcd, you can just see a few pixels of it.

thinkyhead, 2018-11-17:
Yes, you can set it to 0 to disable it in Marlin 2.0.

thinkyhead, 2018-11-17:
As for the LCD issue, I assume you refer to the graphical LCD. I’ll look into it.

closed 2018-11-17 by thinkyhead