MarlinFirmware/Marlin #12456: After Preheat All Menue Options gone

AndreasHeisinger, 2018-11-17:
Hey,

ive installed Skynet Marlin on a Anet A8. 
Everything works fine. 
But when i start to preheat , the Menue is empty. No Options to move Axis, Preheat, Cooldown...
Also when Preheat Temperature is on. 
After Reset the Printer everything is visible. Start Preheat, the same no Menue Options.
Can anybody help me?

Ludy87, 2018-11-17:
Skynet realy? Try Marlin Bugfix-2.0.x

AndreasHeisinger, 2018-11-17:
Thank you! Can i use my configuration.h or should i edit the configuration.h form the bugfix ?

Ludy87, 2018-11-17:
Change the bugfix configuration files (Configuration.h and Configuration_adv.h) to your needs.
The copy would lead to errors.

thinkyhead, 2018-11-17:
We don’t do support for modified forks of Marlin. Please contact the Skynet project.

closed 2018-11-17 by thinkyhead