MarlinFirmware/Marlin #12457: [2.0.X] Underextrusion after Retract: Extruder isnt following gcode

DerEchteLanger, 2018-11-17:
After big issues with underextrusion after retracts im currently researching possible causes. I found out that my extruder isnt doing what its supposed to do in the gcode as repeated backward/forward movements result in the extruder gear "moving backwards" although the sum of those movements should move it forwards. This got realy clear when i tried printing a retraction test simliar to this one: https://www.thingiverse.com/thing:909901

Ive shortened the gcode to the last layers and removed filament (and heater stuff etc.), which resulted in this gcode: https://pastebin.com/T8TyzXjc
The extruder gear should end the movement in a slightly forward position compared to the beginning, but as clearly visible in this video thats not the case: https://streamable.com/j4hx2
You can see how the total movement of the extruder gear is clockwise which results in a net retract.

The issue gets more strange if i clear all X/Y/Z movements from the gcode file, so only the extruder is moving: https://pastebin.com/WW6kBABm
This results in the extruder working as intended, although its moving much faster than combined with the X/Y movement: https://streamable.com/9ruv4
There is no net backwards motion, the extruder ends in approximatly the same position as it starts (its only extruding about 0,2mm during the gcode file).

My Setup: Im using watterott TMC2130 on X/Y/E, driving my Extruder motor with 700mA (no overheating), which is way more than enough since identical motors drive the X/Y axis with identical settings. The Titan is a clone by trianglelabs and has no mechanical slack, everything is sitting tight.

Im running Marlin 2.0.X on an Ardunio/Ramps, my Marlin is about 2 weeks old, although i also had this issues with a version from May. My configs:
Configuration.h: https://pastebin.com/f2Ywt23h
Configuration_adv.h: https://pastebin.com/dJKThiBr
BTW: I tried disabling S-Curve acceleration, it didnt help.

Im using Octoprint in a Raspi 3 to feed the gcode with a baud rate of 250000, sadly i cant test other drivers or printing from a SD card in the next days is i dont have the hardware for that here.

gloomyandy, 2018-11-17:
How do you know that 700mA is enough? The acceleration patterns of the extruder are totally different (partly due to to the much higher steps per mm thanks to the gearing of the titan) to that for the X and Y axis. I'd be very tempted to try increasing the current of the extruder and see if that makes any difference. You might also want to try using spreadcycle rather than stealthchop, there have been a number of reports of lost steps when using TMC drivers for the extruder.

DerEchteLanger, 2018-11-17:
Now i feel a litte bit stupid, since ive spend over a week trying to figure this one out.

Yeah, either raising the current or disabling stealthchop fixes the issue. I never had the idea that an empty (gearded) extruder could allready overburden a stepper which has no problem tossing around a heatbed with the same settings...

Now i have anouther issue, though, since neither higher current nor spread cycle are a sustainable solution for me. If i increase the current my motor gets to hot during longer prints (to hot to touch, i fear it will soften the PETG mounting), if i use spread cycle the noise (high pitch whine) gets unbearable. It means im off finding a new motor for my extruder, thanks for the help.

gloomyandy, 2018-11-17:
I'm no TMC expert but perhaps using hybrid mode could help? I think it switches between stealthchop and spreadcycle and might give you the best of both worlds? But as I said I really don't know enough about it to be sure!

Beherith, 2018-11-18:
When using a TMC2130 in stealthchop on an extruder, be mindful not to exceed a jerk of 10mm/s, and an acceleration of approximately 600ms. 
With these settings, extruder is very reliable and smooth for me. 
I have a regular direct drive extruder.