MarlinFirmware/Marlin #12459: G34 do nothing when enabling Z_STEPPER_AUTO_ALIGN

dough29, 2018-11-17:
I have a Creality CR-10 + MKS Gen L v1.0 + BLTouch (genuine) + Dual Z (one driver per Z motor), everything is working fine (G28, G29, printing and so on...).

You can find my custom config diff from bugfix-2.0.x here : https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...dough29:test-z-auto-align

I enabled dual Z steppers auto-alignment :

`#define Z_STEPPER_AUTO_ALIGN`

I was thinking Y to go to 150 and X to 10 for a first Z probing and then X to 290 for second Z probing and alignment.

`Send: G34
Recv: ok`

Nothing happens.

Am I missing something ?

Thanks for reading me.

dough29, 2018-11-17:
I found how to enable debug output so here is the result, did G28 just before :

`Recv: >>> G34
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X:-41 Y:-10 Z:-1.30 (Left-Front & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR
Recv:  (disabled)
Recv: 
Recv: set_bltouch_deployed(0)
Recv: > probing all positions.
Recv: >>> probe_pt(10.00, 150.00, raise, 0, probe_relative)
Recv:   current_position=(191.00, 160.00, 11.30) :
Recv: > PROBING FAILED!
Recv: <<< G34`

GMagician, 2018-11-17:
>  PROBING FAILED!

it may give such error when position is not reachable. Can't extract your cnf from link so can't say more

dough29, 2018-11-17:
Here are the files :
https://github.com/dough29/Marlin/blob/test-z-auto-align/Marlin/Configuration.h
https://github.com/dough29/Marlin/blob/test-z-auto-align/Marlin/Configuration_adv.h

For Z stepper auto align :

#define Z_STEPPER_AUTO_ALIGN
#define Z_STEPPER_ALIGN_X { 10, 290 }
#define Z_STEPPER_ALIGN_Y { 100, 100 }

I tried with lower values (my printer can easily reach 290) and nothen happens too.

dough29, 2018-11-17:
Send: G34
Recv: >>> G34
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X:-41 Y:-10 Z:-1.30 (Left-Front & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR
Recv:  (disabled)
Recv: 
Recv: set_bltouch_deployed(0)
Recv: > probing all positions.
Recv: >>> probe_pt(10.00, 10.00, raise, 0, probe_relative)
Recv:   current_position=(0.00, 0.00, 10.00) :
Recv: > PROBING FAILED!
Recv: <<< G34
Recv: ok

GMagician, 2018-11-17:
may you try to execute a G92 X0 Y0 post report you get

dough29, 2018-11-17:
I didn't plya with G92, here is the report :

(did G28 XY before)
Send: G92 X0 Y0
Recv: X:0.00 Y:0.00 Z:11.30 E:0.00 Count X:0 Y:0 Z:4520
Recv: ok
Send: G34
Recv: >>> G34
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X:-41 Y:-10 Z:-1.30 (Left-Front & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR
Recv:  (disabled)
Recv: 
Recv: set_bltouch_deployed(0)
Recv: > probing all positions.
Recv: >>> probe_pt(10.00, 10.00, raise, 0, probe_relative)
Recv:   current_position=(0.00, 0.00, 11.30) :
Recv: > PROBING FAILED!
Recv: <<< G34
Recv: ok

gloomyandy, 2018-11-17:
Can your machine get the probe (not the nozzle) to those locations? I suspect that is the problem. You have a fair offset for X which may be causing the issue.

dough29, 2018-11-17:
So now my custom version is up to date and working Z stepper align : https://github.com/dough29/Marlin

Thanks a lot @gloomyandy 👍

closed 2018-11-17 by dough29