MarlinFirmware/Marlin #12460: Help with tmc2130

italocjs, 2018-11-17:
Hello guys, i've been trying to use the tmc2130 for the past two weeks since i bought it, but didnt managed to get a single sucessful print so far.

I bought three tmc2130 and replaced my (working) a4988.  i'm using the driver on standalone mode (soldered jumper), with a 15mm heatsink above it, and a 40mm fan directly above the drivers, so i dont believe overheating is a issue (i can keep my finger at it while working easily)..

I'm getting constant layer shifting on it, here are the things i already tried:
>decrease acceleration to 1000 on every axis
>reduce printing speed to 100mm/s
>reduce movement speed to 150mm/s
>increase current (1.1a rms = 1.7A peak | stepper is 1.7A)
>disable S-Curve

Anyone has any idea on how to fix this? if i put back the a4988 the printer works perfectly!. Also, if possible, i'd like to keep using the standalone mode

Printer kinematics: corexy
Steppers: 4.2kgf - 1.7a  17hs4401
Drivers: TMC2130 on X Y and Z
Firmware: latest 2.0.x (today version)

italocjs, 2018-11-18:
Changing the driver to spreadcycle helped but not solved 100%, im buying two tmc2208 to try before giving up and going back to the a4988

swilkens, 2018-11-19:
Which version of the firmware are you using? How bad are the layer shifts?

There is some activity surrounding layer shift issues in the firmware at the moment:

https://github.com/MarlinFirmware/Marlin/issues/12403

I assume you've looked at things such as belt tension to make sure this isn't a mechanical issue?

gloomyandy, 2018-11-19:
Just be aware that tmc2208 is in stealthchop by default and it is not easy to reconfigure them into spreadcycle, you can't do it via jumpers.

italocjs, 2018-11-19:
> Just be aware that tmc2208 is in stealthchop by default and it is not easy to reconfigure them into spreadcycle, you can't do it via jumpers.

I've disabled the custom configs for tmc2130 on marlin and changed the TMC2130 to standalone as spreadcycle and it fixed the lost steps for me, so i'll keep trying on it, maybe the problem people here are having might be related to the transition between stealthchop and spreadcycle, 

About the 2208, i never used them before, but from what i read, i can change it using uart, just like the tmc2130 changes over spi

italocjs, 2018-11-19:
> There is some activity surrounding layer shift issues in the firmware at the moment:
> 
> #12403
> 
> I assume you've looked at things such as belt tension to make sure this isn't a mechanical issue?

Confirmed, not mechanical issue, when i'm using the A4988 i have no lost steps

gloomyandy, 2018-11-19:
Yes you can change them over uart, but you will probably need a uart per device rather than being able to share one set of pins like you can with spi. There are other ways but that gets more complex.

boelle, 2019-02-21:
@italocjs problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead