MarlinFirmware/Marlin #12463: Not enough RAM Mega 2560

Andreiva13, 2018-11-17:
Hello!
How could i decrease the RAM usage on my configuration ?
TIA
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2592262/Marlin.zip)

Roxy-3D, 2018-11-17:
What are your values?   What do you want to get to?

Bump this down to 7:    #define GRID_MAX_POINTS_X 7
Turn this off:  // #define SDCARD_SORT_ALPHA

Andreiva13, 2018-11-17:
Thanks i will try that tommorow also why if i try to use tmc2130 insted of tmc2130_standalone it gives a error

Roxy-3D, 2018-11-17:
I don't know about the tmc stuff....   Somebody else will need to jump in on that.

RudyBenner, 2018-11-18:
Fix one problem at a time. Unless you like vertigo.

Andreiva13, 2018-11-18:
Yeah true thanks

Andreiva13, 2018-11-18:
Thanks the sorthing thing helped me :)
Now those tmcs

teemuatlut, 2018-11-18:
If you want to report an error, it's usually helpful to describe said error.

Andreiva13, 2018-11-19:
Thanks for the advice but I got it working somehow thanks

closed 2018-11-19 by thinkyhead