MarlinFirmware/Marlin #12464: [bugfix-2.0.x] IDEX mode burns holes in plastic and trips bed overheat protection

vatral, 2018-11-17:
### Description

In the IDEX mode, when the active tool is changed, the new tool heads off to the position of the old one. This results in a number of very undesirable behaviors if the next commands don't make it immediately move.

* The extruder usually ends up right above previously extruded plastic. If the next thing that happens is a temperature change, then the extruder ends up burning a hole in whatever is under it until it comes up to the right temperature.
* If the extruder happens to end up in the vicinity of the bed's heat sensor, it can trip bed overheat protection.
* It's arguably a waste of time, since a lot of dual extruder 3d printing doesn't involve continuing where the previous one left off.

For instance, here's what Cura does for an IDEX printer:

    ; End of work of T0
    G0 F7200 X153.573 Y74.943
    G1 F1200 E45.74184
    G92 E0 
    T1            ; Extruder change, T1 moves over what T0 has just extruded
    G92 E0 
    M105   
    M109 S200     ; T1 is now heating up to 200C, from the previous standby temperature.
    M104 T0 S175  ; T0 is set to the standby temperature
    G1 F1200 E9.5 

### Steps to Reproduce

1. Use IDEX printer
2. Print a two color object with Cura, using a reduced standby temperature for the inactive extruder
3. Observe how the newly active extruder ends up waiting to heat up over previously printed material

**Expected behavior:**

On tool change, the previously active tool should park. The newly active tool should remain in its parked position until commanded to move.

**Actual behavior:**

On tool change, the previously active tool parks. The newly active tool moves over the previous one's position.

Roxy-3D, 2018-11-17:
Can you attach your Configuration.h files?

vatral, 2018-11-18:
Here you go:

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2592399/config.zip)

InsanityAutomation, 2018-11-18:
Full control mode is really designed for optimizing alot of this. Autopark has no idea a temp command is coming next. There's nothing it can do about it. It has no idea a different position is coming next. It sounds like you're expecting the behavior of full control while in auto park mode.

vatral, 2018-11-18:
It could just let the new tool stay parked, until commanded to move. Then the code above would work perfectly fine.

On an IDEX printer there are plenty scenarios in which it makes little sense to move the second carriage over the previous one's position. For instance if using one for supports and another for the object, a lot of the time the movement isn't going to overlap. The support extruder will be building a support pillar separated from everything else that will eventually reach some feature that will need support, for instance.

Also, nothing I can find in Cura seems to indicate it knows anything about full control mode.

InsanityAutomation, 2018-11-18:
The firmware can't assume the next move is a travel move. If it's an extrusion move slicer assumes it's coming from the current position. In the case of soluble supports, sure. In the case of color swaps, maybe not. Full control mode needs tool change scripts written with knowledge of what you're trying to accomplish. Auto park is safe with no knowledge of what's next.

vatral, 2018-11-18:
I see your point about color swaps, but speaking from personal experience, an IDEX printer is an absolutely awful way to do that. The inactive head ends up dripping plastic for some time, so after starting to move again it needs to extrude out some amount to prime itself. And after that, it still needs to move out of the parked position without leaking too much.

Even if returning to the previous position makes sense, a lot of the time an IDEX printer would print nothing at first, and priming itself right over previously printed material isn't likely to work terribly well either.

If you're doing color swaps, it's going to be far less painful to use a single nozzle for it.

My current understanding is that optimal usage of an IDEX setup is something like this:

* Let the inactive extruder cool down to minimize dripping
* Warm up newly active extruder
* Prime extruder to compensate for anything that dripped out
* Retract
* Move

vatral, 2018-11-18:
Here for instance is a video of how the same kind of printer that I have operates:

https://www.youtube.com/watch?v=96zY4V7rfuc#t=3m08s

InsanityAutomation, 2018-11-18:
And in full control mode, a slicer script can do that. However auto park cannot rely on the gcode after tool change to begin with a travel move. An automatic mode must be failsafe, not necessarily the most optimal. A post switch prime is common in s3d with an extra restart distance. I recently added retract and prime control on tool change, and may add an extra prime distance that it can do while parked, which would help the situation, however if you're still cooled off it'll be an issue. Slicer will need to send temp change before calling tool change, and will need to move the nozzle, thus getting closer to what full control needs to take full advantage of it.

As for cooling inactive extruder, on long layers that's fine however if you have a large number of small layers to go between, the time impact would be huge. In theory we could add an automatic cool down to the menus and utilize some of the inactive temp storage code from singlenozzle, however there are quite a few bigger issues to solve first.

InsanityAutomation, 2018-11-18:
If you want to have the process this optimized, I really have to suggest full control mode. You'll need separate profiles for short layers (without cooldown) vs long (with cooldown), color change (return to same position) vs support/infill only (leave parked or return to next position) and variable extra prime amount after temp commands in long usage. Given the multiple scenarios, this really is beyond the scope of what firmware can be expected to figure out automatically.

To do anything better here we would need to look ahead in the buffer and find several future commands, which may not even be in the buffer yet. It would not be reliable.

vatral, 2018-11-18:
I agree that full control mode is of course better. But some programs like Cura (which as far as I know is quite popular) don't seem to know about it.

Then I think there's the question of what's the use case for DXC_AUTO_PARK_MODE? I take it's to try to accommodate slicers that don't know anything about IDEX and have no understanding of what it involves. But as I'm seeing, in practice it results in trouble with temperature and priming.

How's this?

* Add a strong suggestion to enable TOOLCHANGE_FILAMENT_SWAP in the DXC_AUTO_PARK_MODE explanation.
* Perhaps add a TOOLCHANGE_COOLDOWN to go along with it, to do what's Cura trying to do here.
* Explain in the documentation that cooldown should be disabled in the slicer under DXC_AUTO_PARK_MODE.

How does this sound? I'd be willing to try to do it.

I think with with the automatic priming, retraction (and cooldown is also nice), DXC_AUTO_PARK_MODE should be decently useful, even if not optimal. Which I think is the intention here anyway.

InsanityAutomation, 2018-11-18:
I just added the auto retract and prime a couple weeks ago, the suggestion may be added at some point when it's more widly used and confirmed solid on more hardware than just what Roxy and myself have on hand. The cool down may be considered with caution but only with menu options to disable it as for small objects it will be very time consuming.

You are correct that no slicer takes idex modes into account yet. That's why tool change scripts are required to be custom written for full optimization.

There are some more improvements in the works for idex, but in part because of the process optimization capability mentioned, full control is still the suggested default.

InsanityAutomation, 2018-11-18:
Also as for auto park's usefulness, I can slice for a single nozzle (creality cr-x), chimera (Taz dual v3), switching (ultimaker) or idex, and run the identical gcode on every machine. It's a good foolproof starting point for less advanced users to learn on or stable consistent use for production shops.

boelle, 2019-02-21:
@vatral problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead