MarlinFirmware/Marlin #12465: 1.1.9 G33 crash head. (Probe Z-offset)

Cyper77, 2018-11-18:
> in delta calibration>Probe Z-offset item dont work properly. 
because menu item name must be "paper Z-offset" not probe.
my carbon fiber tube broken beacuse of this menu when probe attached.
zprobe works well but movement didnt stop and Bammm to table...
i found explanation in marlin sources. this was very sad to me :(
PAPER ! not a probe...
"G33 P-1" > Check the z-offset with a center probe and paper test.

thinkyhead, 2018-11-20:
The `G33` feature is not well understood by anyone except its author, @LVD-AC. I'm not sure what the meaning of "with a center probe" is, or why the feature would even try to do anything without a probe attached.

LVD-AC, 2018-11-20:
To probe for Z-offset G33 needs to probe (with the probe deployed) and do a paper test (with the probe retracted). G33 does call the right deploy() and retract() routines at the right time, but those do not pause if the probe requires user interaction. Detachable or manual deployable probes are just not supported in Marlin. IMO such a probe type is missing in configuration.h; and the implementation would be in the deploy() and retract() routines to send a message to deploy or retract the probe, and to wait for user interaction.

Cyper77, 2018-11-25:
funny :), i explain movement (for delta Normally Opened(NO) probe attached)
-head moves to home z
-head moves to table
-first touch and retract
-and moves to table again for second touch maybe but full force
-this time software dont bother the probe signal
-crash-bammm-smash etc....

closed 2019-02-19 by thinkyhead