MarlinFirmware/Marlin #12467: [bugfix 2.0.x] error compling

hermes3008, 2018-11-18:
### Description

compiles with this error

In file included from C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu.h:24:0,

                 from C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:31:

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp: In function 'void progress_bar_test()':

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:59:23: error: 'CHARSET_MENU' was not declared in this scope

       LCD_SET_CHARSET(CHARSET_MENU);

                       ^

c:\users\hermes\appdata\local\temp\arduino_build_221521\sketch\src\lcd\ultralcd.h:299:60: note: in definition of macro 'LCD_SET_CHARSET'

           #define LCD_SET_CHARSET(C) set_custom_characters(C)

                                                            ^

c:\users\hermes\appdata\local\temp\arduino_build_221521\sketch\src\lcd\ultralcd.h:299:61: error: 'set_custom_characters' was not declared in this scope

           #define LCD_SET_CHARSET(C) set_custom_characters(C)

                                                             ^

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:59:7: note: in expansion of macro 'LCD_SET_CHARSET'

       LCD_SET_CHARSET(CHARSET_MENU);

       ^

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:66:51: error: 'lcd_moveto' was not declared in this scope

     lcd_moveto((LCD_WIDTH) / 2 - 2, LCD_HEIGHT - 2);

                                                   ^

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:67:35: error: 'lcd_put_u8str' was not declared in this scope

     lcd_put_u8str(int(bar_percent)); lcd_put_wchar('%');

                                   ^

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:67:55: error: 'lcd_put_wchar' was not declared in this scope

     lcd_put_u8str(int(bar_percent)); lcd_put_wchar('%');

                                                       ^

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:68:69: error: 'lcd_draw_progress_bar' was not declared in this scope

     lcd_moveto(0, LCD_HEIGHT - 1); lcd_draw_progress_bar(bar_percent);

                                                                     ^

In file included from C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu.h:24:0,

                 from C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:31:

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp: In function 'void _progress_bar_test()':

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:73:21: error: 'CHARSET_INFO' was not declared in this scope

     LCD_SET_CHARSET(CHARSET_INFO);

                     ^

c:\users\hermes\appdata\local\temp\arduino_build_221521\sketch\src\lcd\ultralcd.h:299:60: note: in definition of macro 'LCD_SET_CHARSET'

           #define LCD_SET_CHARSET(C) set_custom_characters(C)

                                                            ^

c:\users\hermes\appdata\local\temp\arduino_build_221521\sketch\src\lcd\ultralcd.h:299:61: error: 'set_custom_characters' was not declared in this scope

           #define LCD_SET_CHARSET(C) set_custom_characters(C)

                                                             ^

C:\Users\hermes\AppData\Local\Temp\arduino_build_221521\sketch\src\lcd\menu\menu_configuration.cpp:73:5: note: in expansion of macro 'LCD_SET_CHARSET'

     LCD_SET_CHARSET(CHARSET_INFO);

     ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\libraries\LiquidCrystal 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

### Steps to Reproduce

1. copy configs to marlin folder
2. upload scetch via arduino ide 1.8.7 to mks base gen l
3. watch the fun

succesful compile

errored out with above error
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2592641/Configuration.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Ludy87, 2018-11-18:
Declarat a Display!

hermes3008, 2018-11-18:
on configeration.h line 1601

Ludy87, 2018-11-18:
enable ```#define NEWPANEL```

Ludy87, 2018-11-18:
I have created a new PR, please wait for publishing

closed 2018-11-19 by thinkyhead