MarlinFirmware/Marlin #12468: M303 Bed PID Autotune fails

brisma, 2018-11-18:
### Description

Hi, after the latest changes on the M303 GCODE there is a problem with the PID Autotune of the bed. It seems to use the E0 temperature/heater instead of E-1.

### Steps to Reproduce

1. Do a M303 E-1 S60 C6
2. Watch the temperature of Bed and Hotend
3. The autotune of the bed will fails

Or

1. Set the temperature of hotend to 200°C
2. Do a M303 E-1 S60 C6
3. The autotune fails at start with the message "Too hot".

neoxm3, 2018-11-18:
i have exactly the same problem...
 M303 E-1 S60 C6  or  M303 P0 - 1 -2 - 3.... (i tryed all for see what happens)  heats the extruder, and not the bed.
Electronics: Melzi , only one extruder.

RudyBenner, 2018-11-18:
#define PIDTEMPBED

//#define BED_LIMIT_SWITCHING

neoxm3, 2018-11-18:
> 
> 
> #define PIDTEMPBED
> 
> //#define BED_LIMIT_SWITCHING

i have it this way...

gloomyandy, 2018-11-18:
When asking for help please make sure that you upload your configuration files. It also helps if you provide details of exactly which version of Marlin you are using (and if a bugfix build when you downloaded it).

kAdonis, 2018-11-18:
I already submitted a PR #12471  changing byteval() back to intval()
its working for me

ghulands, 2018-11-18:
👍

kAdonis, 2018-11-19:
should be fixed now

brisma, 2018-11-19:
Ok, now is working correctly.

closed 2018-11-19 by brisma

reloxx13, 2019-01-13:
seems not to work, im on latest marlin 2.0.x from today

> Send: M303 E-1 S60 C8 U
> Recv: PID Autotune failed! Bad extruder number
> Recv: ok

Ender 3 and here are my configs:
https://github.com/reloxx13/Marlin/tree/my-marlin/Marlin

PR #12471 (2018-11-19): Fix PID Autotune for Heated Bed
fixes a regression in M303.cpp
e could never be negative so E-1 for PID Autotune the heated bed results in Autotune of the extruder

Related Issues
fixes #12468