MarlinFirmware/Marlin #12475: Issue compiling branch bugfix-2.0x  LPC1768 (MKS SBASE 1.3)

mondobizarro, 2018-11-19:
Hello ?im tryng to compile the **BugFix-2.0x** branch and I'm having the following error:

> Unable to find destination disk.  File must be copied manually.
> Compiling .pioenvs\LPC1768\liba52\30aa480\LiquidTWI2.cpp.o
> .piolibdeps\30aa480\LiquidTWI2.cpp:28:10: fatal error: WProgram.h: No such file or directory

 Here Is the entire compile log:

`Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768; framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x > NXP LPC1768

HARDWARE: LPC1768 100MHz 31.80KB RAM (464KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain+) COMPATIBILITY(strict)
Collected 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.2.2 #cefefbe
|-- <LiquidCrystal> 1.0.0
|-- <30aa480>
|-- <Servo> 1.0.0
 
Unable to find destination disk.  File must be copied manually.
Compiling .pioenvs\LPC1768\liba52\30aa480\LiquidTWI2.cpp.o
.piolibdeps\30aa480\LiquidTWI2.cpp:28:10: fatal error: WProgram.h: No such file or directory
 
******************************************************************
* Looking for WProgram.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WProgram.h"
* Web  > https://platformio.org/lib/search?query=header:WProgram.h
*
******************************************************************

#include "WProgram.h"
^~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs\LPC1768\CMSIS\driver\lpc17xx_wdt.c.o
Compiling .pioenvs\LPC1768\CMSIS\lib\chanfs\ff.c.o
*** [.pioenvs\LPC1768\liba52\30aa480\LiquidTWI2.cpp.o] Error 1
Compiling .pioenvs\LPC1768\CMSIS\lib\chanfs\ffunicode.c.o
C:\users\salas\.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib\chanfs\ff.c: In function 'f_setlabel':
C:\users\salas\.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib\chanfs\ff.c:5061:2: warning: this 'for' clause does not guard... [-Wmisleading-indentation]
for (slen = 0; (UINT)label[slen] >= ' '; slen++) ; /* Get name length */
^~~
C:\users\salas\.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib\chanfs\ff.c:5081:2: note: ...this statement, but the latter is misleadingly indented as if it were guar
ded by the 'for'
{ /* On the FAT/FAT32 volume */
^
 [ERROR] Took 11.74 seconds
 
 [SUMMARY]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment LPC1768                     [ERROR]
 [ERROR] Took 11.74 seconds
 `
Here is the config.h and Config_adv.h files that I'm using.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/2593526/Configs.zip)
The problem appear only using the lastest changes applied to the branch, with my las pull from the repo compiles ok.

Any suggestion?

p3p, 2018-11-19:
> LDF MODES: FINDER(chain+) COMPATIBILITY(strict)

That can't be the recent clone / download?, the LDF has been off for a while now, set in the LPC1768 section of `platformio.ini`.

RudyBenner, 2018-11-19:
Other than not finding the SD card, it compiled normally.
[firmware.zip](https://github.com/MarlinFirmware/Marlin/files/2594502/firmware.zip)
"Unable to find destination disk.  File must be copied manually."

mondobizarro, 2018-11-19:
> 
> 
> > LDF MODES: FINDER(chain+) COMPATIBILITY(strict)
> 
> That can't be the recent clone / download?, the LDF has been off for a while now, set in the LPC1768 section of `platformio.ini`.

Indeed, it was. I was keeping the platformio.ini from the previous branch. Updating this file with the current branch compiles correctly.

Thank you very much and greetings.

I close this issue.

closed 2018-11-19 by mondobizarro