MarlinFirmware/Marlin #12487: [2.0.x, RFC] "Print from SD" as the last menu option

marcio-ao, 2018-11-19:
We modify our firmware to place the "Print from SD" as the very last option of the menu. The reason we do this is because it is a very commonly used option and moving to the bottom can be done with a quick flick of the wrist on the encoder. In Marlin upstream, it is placed in the second spot, presumably to make it accessible, but paradoxically that placement makes the user turn the encoder by a precise amount, which is hard.

I would like to propose that the "Print from SD" be moved to the very bottom. It may seem counter-intuitive to have the most used feature on the bottom, but once you try it, you will love it!

Ludy87, 2018-11-20:
what about the users with button? Click, Click, Click ...

swilkens, 2018-11-20:
With this frequently used option at the bottom of the menu, it might end up out of the visible menu scale if other features are enabled - requiring a user to have to scroll to even see the option.

If you know it's there you might prefer this, but it seems indeed counter intuitive to have a frequently used option out of screen.

InsanityAutomation, 2018-11-20:
For scroll wheels, I agree with marcio. This is also closer to the 1.1.9 position where it's close to the bottom. Tier 1 manufacturing facilities actually have a workload calculation scale and a fine move is considered to take 2.5-3x the tact time as a rough move depending on if a tool is involved.

marcio-ao, 2018-11-20:
> With this frequently used option at the bottom of the menu, it might end up out of the visible menu scale if other features are enabled - requiring a user to have to scroll to even see the option.

@swilkens: There will always be features that are hidden unless the user scrolls, so I'm not sure this is a strong argument against using the most efficient placement for the particular controller. That said, just making it a configuration option and entirely up to the user would be a good compromise.

Ludy87, 2018-11-20:
Or a other idea, the first position🤔

InsanityAutomation, 2018-11-20:
First position is always back / exit so it would be the second, and thats exactly the scenario compared against above. Given some manufacturers issues with noise on the button line causing a ton of clicks, id be concerned changing that behavior.

InsanityAutomation, 2019-01-14:
FYI see https://github.com/MarlinFirmware/Marlin/pull/12893/ for the rest of the menu fixups