MarlinFirmware/Marlin #12489: [bugfix 2.0] Z axis 2 decimal places

dobrzann, 2018-11-20:
hello,

Is it possible to display 2 decimal places on Z axis as it is in marlin 1.1.x ?

eg.
during G29 probing I see only one decimal place (10.0), why?

Roxy-3D, 2018-11-20:
Yes.    And I miss the extra precision on the Z-Axis.    I just haven't gotten around to adding a Configuration_adv.h option to bring it back.    The problem is the very limited space available on the status screen.   Trade offs have to be made.   And most people want the space allocated to other purposes. 

In my case...   because I focus a lot on bed leveling, I want more precision on the Z-Axis.    When I get time, I'll add an option to bring it back.   Or...  If you get impatient....  Please submit a Pull Request with the required code to implement the feature and it will get added to the code base.

> during G29 probing I see only one decimal place (10.0), why?

What the LCD Panel is displaying is not the instantaneous status of the machine.    It is displaying where it is trying to go.   (And if you double check me on this...  I suspect you will see a -10.0 and not a +10.0 on your display.)    The reason is Marlin is assuming it should bump the bed before it gets -10 mm below where the first bump was detected.

teemuatlut, 2018-11-20:
Maybe reserve a certain width of space and adjust the format according to the value.
Like -1.2 -> 1.23 -> 12.3 -> 123

thinkyhead, 2018-11-20:
It actually is drawing two decimal places, but it's getting chopped off on the graphical display. We just need to move Y 3 pixels to the left, and Z 6 pixels to the left.

thinkyhead, 2018-11-20:
ec9d21a234656b41dd934933d42c2aaa2599d06e

closed 2018-11-20 by thinkyhead