MarlinFirmware/Marlin #12492: Using Level Fade Height with AUTO_BED_LEVELING_3POINT

Murilolana, 2018-11-20:
Hi,

I've just setup a new printer that uses a slight different concept of build plate. It's a 8mm acrylic platform with 3 magnets inserted at specific points through the botton. These magnets works as indexers to the platform support as well as the inductive probe sensor points. So, basically it is a removable non metal build plate that works with inductive sensors.
 It is wotking nicely, except that I can't put level fade height to work with it. I've edited the line in the configuration.h as this:

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL) || ENABLED(AUTO_BED_LEVELING_3POINT)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT

The printer measure the 3 points nicely and do the proper leveling through printing, but the level fade height doesn't work.
My start gcode is this:

G28 ; home all axes
G29; Bed level
M420 C1 S1 Z6

Expected behavior:
- Heat up,
- Home all axes,
- Do probing,
- Fade compensation untill Z6 and stop level compensation

Actual behavior:
- Heat up,
- Home all axes,
- Do probing,
- Do level compensation on the entire part height

So, is there a way to put this to work?

Thanks a lot.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-11-20:
Fade is only for mesh-based leveling, and not for 3-point leveling.

closed 2018-11-20 by thinkyhead

Murilolana, 2018-11-20:
> Fade is only for mesh-based leveling, and not for 3-point leveling.

Could it be implemented?

thinkyhead, 2018-11-20:
No.