MarlinFirmware/Marlin #12494: m600 function not working

ondrax1, 2018-11-20:
Hello, i have question. I have Anycubic with derhopp marlin 1.1.x  I update original FW, becouse i need function m600 and changing filament. But i have problem with this. I setup 
#define ADVANCED_PAUSE_FEATURE

and 

#define NOZZLE_PARK_FEATURE

After check, arduino ide wrote me     #error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller or EMERGENCY_PARSER." And i setup function Emergency parser. After that i flash this FW to my printer. Pausing is working! (fantastic) but i dont know how i can continue print, after change filament. If i try make pause, temperatures goes to 0 and pring progress is stopped.

video here: https://youtu.be/RSgZyg1jOtM

(link here https://github.com/derhopp/Marlin-with-Anycubic-i3-Mega-TFT)

Sorry for my bad english.

Thank you so much

thinkyhead, 2018-11-22:
To continue without an LCD you have to send `M108`.

ondrax1, 2018-11-22:
Hello, i use pronter face. M600 in gcode make me pause, i change filament. Send m108 via ptonter face and problem
[image]

enable "NOZZLE_PARK_FEATURE" (Configuration.h), "ADVANCED_PAUSE_FEATURE" (Configuration_adv.h) and "EMERGENCY_PARSER" (Configuration_adv.h).

thinkyhead, 2018-11-27:
Try sending `M108` two or more times.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead