MarlinFirmware/Marlin #12496: Interfacing with closed loop stepper

alexxy, 2018-11-20:
What is the best way to interface with closed loop stepper motor?

Currently I working on simple design for closed loop stepper comntroller that can be very cheap but will its work. So basicaly it is stm32 mcu with tmc motor driver and magnetic encoder. So the main question is interface between control board and comntroller. Ideas?

comps, 2018-11-20:
See #11772 and the discussion in it.

alexxy, 2018-11-20:
Good point. But idea is to make motor controller board to all of position control thing.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

alexxy, 2019-04-22:
Its not user support question. I'm developing board similar to nano zero or machaduino. So I rise question about interfacing with Marlin.

So possibilityes are:

- step/dir/enable interface + additional pin to report that movement done
- step/dir/enable interface + spi or i2c
- can bus to rule em all
- i2c to controll everything
- spi to control everything

Or any other suggestions? I think in case of using closed-loop drives marlin need some kind of feedback to  know if movement finished or not, or it get an error.

I tested encoder resolution and stability: its ~0.02 degree (for example for 200 steps/rev stepper with 16 microstep we need ~0.1125 degree, so resolution is enough for 64-128 microsteps... )

alexxy, 2019-04-24:
I now implementing interface in  that way
1. closed-loop board exposes step, dir, enable pins that are translated into step dir enable of driver
2. closed-loop board exposes additional two pins: ready and error that are used for reporting status
3. closed-loop board exposes spi for driver configuration (as well as uart)

crazyp1g, 2019-05-17:
@alexxy   
Why not consider like that TMC 4361 and TMC5160

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

InsanityAutomation, 2019-06-22:
No, there is still plenty to be done around closed loop steppers. I just got a couple in 2 days ago to start playing with. The TMC motion controllers above would be hugely beneficial as well.