MarlinFirmware/Marlin #12498: error checking

manotroll, 2018-11-21:
I'm ordering and checking this error.
In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\MarlinConfig.h:32:0,

                 from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\SdFatUtil.cpp:29:

Configuration_adv.h:1654:2: error: #endif without #if

 #endif // CONFIGURATION_ADV_H

  ^

In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\MarlinConfig.h:32:0,

                 from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\I2CPositionEncoder.cpp:31:

Configuration_adv.h:1654:2: error: #endif without #if

 #endif // CONFIGURATION_ADV_H

  ^

In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\MarlinConfig.h:32:0,

                 from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\SdFile.cpp:29:

Configuration_adv.h:1654:2: error: #endif without #if

 #endif // CONFIGURATION_ADV_H

  ^

In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\MarlinConfig.h:32:0,

                 from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\leds.cpp:27:

Configuration_adv.h:1654:2: error: #endif without #if

 #endif // CONFIGURATION_ADV_H

  ^

In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\MarlinConfig.h:32:0,

                 from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\M100_Free_Mem_Chk.cpp:49:

Configuration_adv.h:1654:2: error: #endif without #if

 #endif // CONFIGURATION_ADV_H

  ^

In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\MarlinConfig.h:32:0,

                 from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\digipot_mcp4018.cpp:23:

Configuration_adv.h:1654:2: error: #endif without #if

 #endif // CONFIGURATION_ADV_H

  ^

In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\MarlinConfig.h:32:0,

                 from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\SdBaseFile.cpp:30:

Configuration_adv.h:1654:2: error: #endif without #if

 #endif // CONFIGURATION_ADV_H

  ^

exit status 1
In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\MarlinConfig.h:32:0,

                 from C:\Users\manot\AppData\Local\Temp\arduino_build_737498\sketch\cardreader.cpp:23:

Configuration_adv.h:1654:2: error: #endif without #if

 #endif // CONFIGURATION_ADV_H

  ^

#endif without #if

my idea is to use mks gen 1.4 and TFT28 2.8 MKS
and as it uses serial disabled everything that is referring to the sd and lcd card to save memory and leave free to other more important things
but I do not know why this error is not giving the clue of what can be

Ludy87, 2018-11-21:
Your Configuration_adv.H is corrupt, load Marlin new

Ludy87, 2018-11-21:
Yes

thinkyhead, 2018-11-22:
Try removing the line it complains about:
```cpp
#endif // CONFIGURATION_ADV_H
```
The new configuration doesn't use…
```cpp
#ifndef CONFIGURATION_ADV_H
#define CONFIGURATION_ADV_H

. . .

#endif // CONFIGURATION_ADV_H
```
Instead we just have this at the top:
```cpp
#pragma once
```

manotroll, 2018-11-23:
tomorrow I will try to record on the board but the error is gone

closed 2018-11-23 by thinkyhead