MarlinFirmware/Marlin #12499: [UI] Make cold extrusion prevention warning more intrusive

lf-, 2018-11-21:
### Description

Cold extrusion prevention is a great idea and I'm sure it's prevented much filament grinding, but it is very confusing to users who are trying to test their extruders or print with ultra-low-temp filaments. The reason for this is that the warning doesn't appear in the jog view they are using and seems to only sometimes appear in the bottom of the screen on the status page (as it could be quickly replaced with some other message).

This is compounded by the fact that fewer and fewer people initially set up their printers using a serial connection, meaning that they might not see the error message sent out over serial.

I've personally assisted multiple users who thought their mainboards were fried because they weren't aware they were running up against cold extrusion prevention in the middle of troubleshooting some other issue or simply testing their extruder. As such, it really needs to bring up a full screen message that can't be accidentally skipped by trying to jog the printer. A simple way to do this is to kill the printer if it occurs, but the initial implementation of it didn't do this for some reason, suggesting there may be some reason not to do so?

lf-, 2018-11-21:
And here's an example from today possibly caused by this: https://www.reddit.com/r/3Dprinting/comments/9yxqkk/error_with_extruder_dont_know_what_broke/

RudyBenner, 2018-11-21:
Not really a firmware issue.

lf-, 2018-11-21:
This is a firmware issue. Users don't see that their extruder is being stopped from moving *by a firmware feature* because the error is invisible.

RudyBenner, 2018-11-21:
[image]
Its up to the host to react. Or just turn off the feature.

lf-, 2018-11-21:
Perhaps it is up to the host to react-except for the inconvenient fact of
none of the hosts reacting and many, many people setting up their printers
and testing with no host attached to begin with.

Why should the user jogging an axis using the screen on the printer see an
error message on their computer 5m away but not where they're actually
using the user interface?

InsanityAutomation, 2018-11-21:
I'd think a simple beep, or hide the extruder jog menu when too cold would be more than sufficient. Maybe change the message based on too cold or not, so instead of Extruder, you just get Ext. Too Cold then when it heats message goes back. That's the limit of what we could really do imo.

lf-, 2018-11-21:
Well, of those options, replacing the menu with a button which does nothing and says "too cold" would be good. OctoPrint still has bad UI with regards to this assuming it won't just disconnect itself upon getting the message (I'm not sure if this is the case), but fixing the menu on the printer's screen not to be there is a good start. 

Why should errors when not connected to a host *not* be fatal actually?

InsanityAutomation, 2018-11-21:
Call tool change with retract / prime on, and it'll trigger. A few other areas. It's good for simulation as well. This is more of a warning than a hard error.

Ludy87, 2018-11-22:
if you care so, add this line to planner.cpp

```c++
ui.set_status (MSG_HOTEND_TOO_COLD);
```

under the line `SERIAL_ECHOLNPGM (MSG_ERR_COLD_EXTRUDE_STOP);`

hamster65, 2018-11-23:
Is there a way to display non-critical / non blocking error messages like "prevented cold extrutions" on lcds? That could be handy for cases like this.

thinkyhead, 2018-11-23:
All alert messages block general info messages. So if a "too cold" message was presented as an alert, it would then block general informational messages from then on. If it was presented as a general info message, then it would remain on-screen until the next general info message was presented. Since general info messages can be configured to expire, that may be the best option.

lf-, 2018-11-23:
But that's still on the info screen, which is not right in front of the
confused user who is wondering why their extruder isn't turning, and
wouldn't think to look at the info screen.

Actually, would it be possible to show those info messages on the jog
screen as well if hiding the jog isn't possible?

thinkyhead, 2018-11-23:
I see. If someone wants to implement that, please do. I'm quite busy with other features right now.

thinkyhead, 2018-11-23:
Note that at the moment, in Marlin 2.0 you cannot even select E movement unless the hotend is heated.

lf-, 2018-11-23:
So this is actually already fixed in a pretty good way. I'll close this, as it is just a matter of waiting for the proliferation of 2.0.

closed 2018-11-23 by lf-