MarlinFirmware/Marlin #12501: Digole 160X128 Color OLED support (Serial/I2C/SPI)

vic10us, 2018-11-21:
Is there support for Digole 160X128 Color OLED support (Serial/I2C/SPI)? If not, is there an API or Interface I can implement to add support?

ejtagle, 2018-11-22:
Right now, Marlin only supports 128x64 monochrome displays through u8glib.
Probably it could be done, but the different resolution, and no color information on graphics would make it less useful.
The other choice would mean to redo all the UI

vic10us, 2018-11-22:
Would scaling be an option? I don't know how the write functions are implemented but it can't be that difficult.

ejtagle, 2018-11-22:
Essentially, you would need to reimplement u8glib on that display, while doing upscaling on the fly... - Doable... Probably...

AnHardt, 2018-11-22:
But even then ugly with a different aspect ratio and broken  scaling factor in x.
For a 320x240 ILI9341 i once tried to scale Xx2,Yx3 and using the remaining space for touch buttons -> too ugly! Fonts unreadable.

thinkyhead, 2018-11-22:
I would suggest that you make everything as a standalone sketch as a first step, and implement the composition part as monochrome bitmaps, colorizing them as they're sent over to the display. This would help with storage, since there's not a lot of memory on the average Arduino.

Then, once you have screen composition working well, you could implement it as an entirely separate peripheral, ignoring all the code inside of Marlin's `lcd` folder. Just try to make a pretty status screen with it as a first step.

Once you've got it working at the most basic level, then we could look at what it would take to integrate things like menus and value editing with an added encoder wheel. But the first step is definitely to get it working at the most basic level first.

Does it need to follow the same pattern as the u8g library? It would definitely help if the interfaces you build match the ones we use in Marlin for drawing text, lines, and bitmaps. That would make it easier to integrate.

Is it a worthwhile display to use? Maybe. We'll have to see how it looks and performs.

ejtagle, 2018-11-22:
I am working on a ILI9331/9441/9841 version of the UI (as you said, pretty hard on AVR, but results are VERY promising so far.
But 160x64 would probably require a full rewrite, as vertically, is is nearly like the ST7290, so layout probably could be the same, maybe a little more information on the status screen...

marcio-ao, 2018-11-26:
> If not, is there an API or Interface I can implement to add support?

One option would be to implement it as an EXTENSIBLE_UI (see files in "Marlin/src/lcd/extensible_ui"). But taking that approach would require you to build an entire UI from scratch, since that API isn't meant to extend the default menu+encoder scheme used in Marlin.

boelle, 2019-02-21:
@vic10us still relavant? or has problem been solved?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead