MarlinFirmware/Marlin #12502: BLTouch Probe Offset not holding

NXSmiggy, 2018-11-21:
Hi All,

Having an issue getting the probe offset to work, seems that after a `G29 `it resets to Z0 and wont hold the probe offset resulting in me having a 1.6mm gap between the bed and nozzle.

When running the following setup from this thread #7913

1. `M502 `to revert to defaults (and deactivate bed levelling)
2. `M500 `save defaults
3. `M503 `to print out settings. Notice the `M851 `Z probe offset line. Whatever you can read here is what is set as probe offset in the configuration.h of the Marlin firmware.
4. `M851 `Z0 will reset this to 0 for the following calibration
5. `G28 `to let it home in all axis. The nozzle will be above the bed by the distance of the probe trigger point to the nozzle, plus the safety margin configured in the firmware to move down after homing
6. `M211 S0` deactivate software endstops, so we can go past Z0.0
7. `G0 Z0` to go to the probe trigger point (in case there was a safety margin)
8. `M119 `reports the endstops/probe status. Ensure that it reads TRIGGERED for the z-probe. Move up and down with `G0 Z0.05` or G0 Z-0.05 to check that you are in deed right at the trigger point where it switches back and forth between TRIGGERED and open
9. `M114 `note down the exact z height
10. Use G0 Z-x.xx to slowly approach the perfect nozzle height with the paper method. See notes to temperature below.
11. `M114 `Note down the final height, subtract the initial height of the trigger point.
12. `M851 `Z-x.yz Set the calculated z offset (negative, if nozzle is above trigger point)
13. `M500 `Save the value
14. `G29 `start bed leveling (details may vary depending on your bed level algorithm)
15. `M500 `save bed level data
16. `M211 S1` re-enable software endstops

It works perfectly until I try and print, then fails to level below the trigger level of the BLTouch. Starup GCode below

> G21        ;metric values
> G90        ;absolute positioning
> M82        ;set extruder to absolute mode
> M107       ;start with the fan off
> G28; Home All
> G29; Auto leveling
> G1 Z15.0 F3600 ;move the platform down 15mm
> G92 E0                  ;zero the extruded length
> G1 F200 E6              ;extrude 6mm of feed stock
> G92 E0                  ;zero the extruded length again
> G1 F3600
> M117 Printing...

AnHardt, 2018-11-21:
If you are homing with the probe, do a G28 in between 13. and 14. to get the new offset in use. Else you have homed with the old offset and probed with the new.

RudyBenner, 2018-11-21:
You do have EEPROM defined?

RudyBenner, 2018-11-21:
as above following your routine, if you enter M851, it should echo the current value.

[image]

that value should hold

RudyBenner, 2018-11-21:
https://3dprint.wiki/reprap/anet/a8/improvement/autobedleveling

NXSmiggy, 2018-11-22:
Sorted now, thanks for the assist

closed 2018-11-22 by NXSmiggy

warzyd, 2018-12-31:
Hi guys,

I did everything just like above but still it does not print as i wish. When i type G0 Z0 in console, the position of Z0 is -0.6, exactly as i set it up with the command M851. However, if i start printing, nozzle is a little bit up and default Ender 3 screen, shows that Z is 3 while it should be 0 as i already set Z0 and it works as i mentioned.

Thanks in advance!

RudyBenner, 2019-01-01:
I thought this was closed?

warzyd, 2019-01-01:
Yes, it was. But i thought that someone could maybe help me as I am pulling my hair out since few days still trying to fix the same thing.

 Best regards,
Albert

thinkyhead, 2019-01-03:
Best way is to start a new issue and fill in the template so we can open a proper investigation.