MarlinFirmware/Marlin #12509: (FR) Stop and Save vs print loss recovery

just-jason, 2018-11-22:
I have been testing the power loss recovery feature, and I am very happy with the feature. There is one point where I am having a tough time with. When there is a power loss the extruder still has an amount of residual heat, enough to let a couple of mm of filament seep out and leave a blob on the print. When resuming from this state, the blob is cold, hard and in the printing path, this is not ideal for the nozzle or print. This is still a great function to rescue a print in the event of an unexpected power loss. What I would like to see is the option on the LCD to save and stop the print so that the machine can be switched off after saving and resumed after switching back on. This would mean that it would be possible to switch off the machine eg when the lab closes/ you don't feel comfortable leaving the machine unattended and resume your print at a later time, the difference being a clean stop of the print without the blob of extra filament that remains on the print when using the current power loss recovery.

thinkyhead, 2018-11-27:
You can try using the Pause feature and then switching off the machine. I'll add code to save the current state when Pause is invoked.

just-jason, 2018-11-29:
Thanks, that would be great. Could you implement it so that it "fakes" the POWER LOSS ROVERY detect power loss on pin 44, ie it saves its current state to the EEPROM only on pause, not constantly ?

j54n1n, 2018-12-02:
Hello,
I am doing with @just-jason some work on the firmware of a PowerWASP printer that uses a very old version of Marlin. We got the latest release of Marlin up and running on the machine. Digging arround in the old and new firmware we found some bits and pieces that could be interesting for this feature.

On the old PowerWASP firmware they had already a "Stop & Save" feature implemented. So maybe we can try to get it ported to the current Marlin firmware.

The original firmware applies following logic:

If the machine is printing from SD card enable a menu item where you can *stop and save* the print (right below the *pause* menu item).
See also:
 - [Stop and save menu item with reference to lcd_sdcard_stop_save()](https://github.com/unibz-makerspace/POWERWasp/blob/v2.1/DWPOWERWASPF21_Rev2/ultralcd.cpp#L317)

In case of selecting the menu item the print gets *paused* and *cooled down*. It uses then https://github.com/MarlinFirmware/Marlin/blob/261c6f4b96541435de786ddbafd491bc2a9d3bc6/Marlin/src/sd/cardreader.cpp#L629 with the flag set to *true* to save a *resurrection file* with the name **resurr.g** with the info about the current *position* and *settings* of the machine. From there onwards you can shut down and power up the machine and pick up the file again by selecting it on the LCD from the file list and start again the print.
See also:
- [The additional lcd_sdcard_stop_save() function](https://github.com/unibz-makerspace/POWERWasp/blob/v2.1/DWPOWERWASPF21_Rev2/ultralcd.cpp#L279..L287)
 - [The modified CardReader::closefile() method](https://github.com/unibz-makerspace/POWERWasp/blob/v2.1/DWPOWERWASPF21_Rev2/cardreader.cpp#L616..L751)

The only other change needed to make this code work is in the CardReader::openFile() method where they create and/or truncate the file of **resurr.g** when you start new a SD card print.
See also:
 - [The modified CardReader::openFile() method](https://github.com/unibz-makerspace/POWERWasp/blob/v2.1/DWPOWERWASPF21_Rev2/cardreader.cpp#L253..L457)

My thinking about this feature would be to apply mostly the same logic from the old firmware and combine it with POWER_LOSS_RECOVERY feature. Maybe we even can reuse the **BIN** file that gets created from the power loss feature? So we could get also the the autoselection of the file when the machine gets powered on.

What are your thoughts about it?
Feel free to browse arround in the old firmware code: https://github.com/unibz-makerspace/POWERWasp

j54n1n, 2018-12-02:
> 
> 
> Thanks, that would be great. Could you implement it so that it "fakes" the POWER LOSS ROVERY detect power loss on pin 44, ie it saves its current state to the EEPROM only on pause, not constantly ?

I think @just-jason was thinking about saving each command on the SD card instead of the EEPROM. 
See also: https://github.com/MarlinFirmware/Marlin/blob/261c6f4b96541435de786ddbafd491bc2a9d3bc6/Marlin/src/feature/power_loss_recovery.h#L32

With this option enabled it causes of course wear on the SD card but you will need to enabled it when you don't have the pin 44 available for power loss detection (that is valid for our case on the PowerWASP).

Cheers

closed 2019-07-11 by shitcreek