MarlinFirmware/Marlin #12510: X Home always in negative

alkduss, 2018-11-22:
Hello, I have a HBot DYI 3D printer and I have Marlin installed. The problem is that after homming, LCD report X-150, Y0 and Z0. X always show negative value and always I use LCD command for reset this value.
My machine is a HBot 3D printer DYI and I have X, Y and Z home probes at MIN and it works correctly, except for the message that appears later "X-150". What I can do? Very Thanks!!!

alkduss, 2018-11-22:
I have reviewed **Configuration.h** several times and can not find any value that is set to -150 or **X_BED_SIZE** or **X_MIN_POS** or **X_MAX_POS**. In the Marlin I find everything correctly. Very Thanks

swilkens, 2018-11-22:
Please post your configuration.h and configuration_adv.h as well as the output of `M503`  if possible.

Zip the files and attach them here.

Ludy87, 2018-11-22:
Load your configuration.h and configuration_avd.h to GitHub

alkduss, 2018-11-22:
OK guys, as soon as I can upload the files and the result of the **M503** command for your valuable help. Very, very thanks!!!

alkduss, 2018-11-23:
Hi guys, reviewing the **M503** command, I realized that the **M206** command applies the offsets to 0 where the hotend is located. I executed it and now it is working correctly, however I send the documents for review in case they think I should adjust something else.

[COREXY.zip](https://github.com/MarlinFirmware/Marlin/files/2609644/COREXY.zip)

swilkens, 2018-11-24:
Great, if your issue is now fixed remember to close the issue report 😉

thinkyhead, 2018-11-26:
M502, M500 is always advised as a first troubleshooting step.

closed 2018-11-26 by thinkyhead

alkduss, 2018-11-29:
Sorry for the delay, the problem was solved with the information that shows the M503 command, **Thanks to everyone**

alkduss, 2018-11-29:
Oh. I'm sorry... I had already answered