MarlinFirmware/Marlin #12514: How can I add a function (only for the printer I am working on for the moment) that allows me to print all the files on the sd card one after another? I am most interested in the automatic next file execution feature. I noticed the AUTOSTART but if I understood correctly, does it mean that if I rename my 3 files as AUTO0.g AUTO1.g AUTO2.g , they will be processed like a queue after the completion of the previous one?

Yahyaoui94, 2018-11-23:
_Originally posted by @Yahyaoui94 in https://github.com/MarlinFirmware/Marlin/issue_comments#issuecomment-441138829_

Ludy87, 2018-11-23:
i haven't tried - try
» disable your Display and enable `SDSUPPORT`
or
» enable `SDSUPPORT `and declarat `SD_DETECT_PIN`

thinkyhead, 2018-11-23:
You can make a single file that contains lots of [`M32` commands](http://marlinfw.org/docs/gcode/M032.html).

closed 2018-11-23 by thinkyhead

xddarko, 2018-11-24:
> > does it mean that if I rename my 3 files as AUTO0.g, AUTO1.g, AUTO2.g , they will be processed like a queue after the completion of the previous one?
> 
> Yes.
> 
> You can also make a single file that contains lots of [`M32` commands](http://marlinfw.org/docs/gcode/M032.html).

Thank you :)  @thinkyhead  renaming should do it for the moment but I am considering to modify the firmware by adding a file.openNextFile() function but am having trouble finding where to insert it or in which tab precisely. Am considering 4 possibilities: 1-Ultralcd.cpp ; 2-setup() ; 3-Uncomment #define MENU_ADDAUTOSTART ; 4-Marlin.cpp

thinkyhead, 2018-11-24:
You'll need to be very careful about what constitutes "next file," since SD cards don't organize files by name, but only roughly by creation date, and that order is lost when files are deleted and new files are added. So you would need to piggyback off the `SDCARD_SORT_ALPHA` feature. You must also consider how files are opened and started, as this determines what the current "working directory" will be. The `CardReader` class is not very well designed, so finding your way around can be a chore. But if you follow the breadcrumbs in the code you should be able to figure it out. Best of luck!

thinkyhead, 2018-11-25:
> in which tab or function do you think should I place it?

I think you'll find that the implementation will require changing code in various places.