MarlinFirmware/Marlin #12515: [bugfix-2.0.x] 5 hotend setup, T4 PWM pin not functional

paukstelis, 2018-11-23:
I have a custom setup that uses 5 hotends. I am using PWM from various pins on a RAMPS 1.4 board to drive mosfets on a secondary board. Everything works as expected for T0-T3 (hotends 1-4), but no matter what pin I assign to drive T4, it fails. I've confirmed it is not my circuit (swapping connections for T3 and T4 makes T4 work and T3 fail). I've tested on 3 different PWM pins and the result is identical. Measuring output voltage from the pin after setting the temperature to 200C shows that it is indeed 5V volts when not connected to the mosfet, but when the  pin is connected to the mosfet it reads 1.6V and is not driving the FET. I believe this is a Marlin problem because it happens to whatever pin is assigned to T4. The behaviour suggests that the pin is somehow not being assigned properly as an output. Any suggestions on where to look toward correcting this? Thanks.

boelle, 2019-02-21:
@paukstelis problem solved?

paukstelis, 2019-02-21:
Yes. It got fixed after I poked some people on twitter :). Closing.

closed 2019-02-21 by paukstelis