MarlinFirmware/Marlin #12516: Marlin/src/sd/cardreader.cpp: if enabled AUTO_REPORT_SD_STATUS build failed

alexxy, 2018-11-23:
If you compile with AUTO_REPORT_SD_STATUS you'll get compile error because block defined two times

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/sd/cardreader.cpp#L118
and

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/sd/cardreader.cpp#L1053

So what instance of this is real one?

alexxy, 2018-11-23:
Seems like related to 66580f32c2f4d39a82b77e4b6982e7ea6dde0766

Ludy87, 2018-11-23:
remove Line 118-124 and try again

https://github.com/MarlinFirmware/Marlin/blob/fd2998efd5fb888eb1deb432c6790c3d506060c0/Marlin/src/sd/cardreader.cpp#L118-L124

alexxy, 2018-11-23:
I already did it. Just not sure what instance supossed to be main one

closed 2018-11-24 by thinkyhead