MarlinFirmware/Marlin #12517: Using WAIT_FOR_BED_HEATER causes problems: Missing busy messages

FanDjango, 2018-11-23:
### Description

Using **WAIT_FOR_BED_HEATER** and then probing causes movement to wait while the bed heats up correctly as expected (in case the bed was not at the set temperature). 

**But** while this heating is taking place, the busy messages that normally appear are disabled, and although there is a single **"Waiting for bed heating message..."**, the printer remains silent from then on until heating is complete.

This causes a problem with OctoPrint (communication timeout).

Here is the define:
```
#define PROBING_HEATERS_OFF       // Turn heaters off when probing
#if ENABLED(PROBING_HEATERS_OFF)
  #define WAIT_FOR_BED_HEATER     // Wait for bed to heat back up between probes (to improve accuracy)
#endif
#define PROBING_FANS_OFF          // Turn fans off when probing §§§§
//#define PROBING_STEPPERS_OFF      // Turn steppers off (unless needed to hold position) when probing
#define DELAY_BEFORE_PROBING 200  // (ms) To prevent vibrations from triggering piezo sensors
```

It seems that the setting of **BUSY_WHILE_HEATING** is not respected in this case:

```
//
// Host Keepalive
//
// When enabled Marlin will send a busy status message to the host
// every couple of seconds when it can't accept commands.
//
#define HOST_KEEPALIVE_FEATURE        // Disable this if your host doesn't like keepalive messages
#define DEFAULT_KEEPALIVE_INTERVAL 2  // Number of seconds between "busy" messages. Set with M113.
#define BUSY_WHILE_HEATING            // Some hosts require "busy" messages even during heating
```

### Steps to Reproduce

1. Configure as described above
2. Set a Bed Temperature that will take some time to reach
3. Do a G28 Z (of course, you might need a G28 XY first).
4. Note how there are no more busy messages until the bed is heated. Only on "Waiting for bed heating..." message.

**Expected behavior:**

Regular Busy messages every 2 seconds (as configured) while the bed is heating

**Actual behavior:** [What actually happens]

No busy messages appear while the bed is heating

#### Additional Information

You can see the sequence of events in this log excerpt:

```
2018-11-23 13:14:10,268 - Send: N6 M140 S65*80
2018-11-23 13:14:10,272 - Recv: ok
2018-11-23 13:14:10,273 - Send: N7 M104 S150*102
2018-11-23 13:14:10,277 - Recv: ok
2018-11-23 13:14:10,278 - Send: N8 G1 Z+25 F480*44
2018-11-23 13:14:10,283 - Recv: ok
2018-11-23 13:14:10,286 - Send: N9 G28 X Y*27
2018-11-23 13:14:12,295 - Recv: echo:busy: processing
2018-11-23 13:14:13,211 - Recv:  T:102.96 /150.00 B:40.19 /65.00 @:127 B@:127
2018-11-23 13:14:15,641 - Recv: echo:busy: processing
2018-11-23 13:14:17,641 - Recv: echo:busy: processing
2018-11-23 13:14:18,210 - Recv:  T:108.96 /150.00 B:40.86 /65.00 @:127 B@:127
2018-11-23 13:14:19,655 - Recv: echo:busy: processing
2018-11-23 13:14:21,517 - Recv: X:-14.00 Y:-3.00 Z:25.00 E:0.00 Count X:-2800 Y:-600 Z:20000
2018-11-23 13:14:21,525 - Recv: ok
2018-11-23 13:14:21,528 - Send: N10 G28 Z*88
2018-11-23 13:14:23,658 - Recv: echo:busy: processing
2018-11-23 13:14:23,665 - Recv:  T:121.52 /150.00 B:41.77 /65.00 @:127 B@:127
2018-11-23 13:14:25,864 - Recv: Wait for bed heating...
2018-11-23 13:14:43,886 - No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-11-23 13:14:43,892 - Changing monitoring state from "Printing" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
2018-11-23 13:14:43,897 - Connection closed, closing down monitor
```

FanDjango, 2018-11-23:
I've looked at the code:
For example in **probe.cpp**:
```
  #if HAS_HEATED_BED && ENABLED(WAIT_FOR_BED_HEATER)
    // Wait for bed to heat back up between probing points
    if (thermalManager.isHeatingBed()) {
      serialprintPGM(msg_wait_for_bed_heating);
      LCD_MESSAGEPGM(MSG_BED_HEATING);
      while (thermalManager.isHeatingBed()) safe_delay(200);
      ui.reset_status();
    }
  #endif
```
This isn't really geared to allow BUSY messages to still keep on coming while waiting for the bed.
And changing that would seem a bit more involved.
What about producing some BUSY messages in there while waiting if BUSY protocol is activated?

thinkyhead, 2018-11-24:
Try this:
```cpp
if (thermalManager.isHeatingBed()) {
  serialprintPGM(msg_wait_for_bed_heating);
  LCD_MESSAGEPGM(MSG_BED_HEATING);
  thermalManager.wait_for_bed();
  ui.reset_status();
}
```

closed 2018-11-24 by thinkyhead

FanDjango, 2018-11-24:
Won't compile. Needs the following addional commit from you:

Needs one boolean argument both from probe.cpp and motion.cpp

  thermalManager.wait_for_bed(**false**);

Then it will work, I have tested and can confirm: Busy Messages appear, Temperature Messages appear