MarlinFirmware/Marlin #12518: SBASE missing?

RudyBenner, 2018-11-23:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

RudyBenner, 2018-11-23:
I no longer see mention of SBASE on marlinfw.org? 
Have they been banished?

Ludy87, 2018-11-23:
Look here https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/README.md
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/config/examples/Mks/Sbase

RudyBenner, 2018-11-23:
Yes, that last link, show me where SBASE is please.
http://marlinfw.org/docs/development/boards.html

Ludy87, 2018-11-23:
Sorry wrong link

RudyBenner, 2018-11-23:
so NOT in marlinfw.org

closed 2018-11-23 by RudyBenner

thinkyhead, 2018-11-23:
> so NOT in marlinfw.org

This board list is maintained in a Google Doc:
http://marlinfw.org/docs/configuration/configuration.html#motherboard